inthehand/32feet #297: Windows conversion back and forth from Windows native platform to 32Feet device

ThePapaG, 2023-05-03:
Hi,

I am using this library to simplify a lot of the interaction with the windows BLE layer as I found the windows native layer to be difficult. I have opted to use the windows `Windows.Devices.Enumeration.DeviceWatcher` class as my device scanner as it gives me event based operation so that I can event based add, remove, and update a list so that I can operate on that list as it changes.

This is requiring me to convert a windows `Windows.Devices.Enumeration.DeviceInformation` to a `InThehand.Bluetooth.BluetoothDevice` object. I am able to achieve this with `BluetoothDevice.FromIdAsync(id)`, however I am unable to go back.
```csharp
// On add we want to check that we don't have this device yet and then add it
            watcher.Added += async (DeviceWatcher sender, DeviceInformation devInfo) =>
            {
                BluetoothDevice dev = null;

                // lock the list while we operate on it
                lock (devices)
                {
                    // Find the device in the list
                    dev = devices.FirstOrDefault(d => d.Id.Equals(devInfo.Id) || d.Name.Equals(devInfo.Name));
                }

                // If the device doesn't exist then add it
                if (dev == null)
                {
                    // Get the ble device for this id
                    BluetoothDevice bleDevice = await BluetoothDevice.FromIdAsync(devInfo.Id);

                    // lock the list while we operate on it
                    lock (devices)
                    {
                        devices.Add(bleDevice);
                    }
                }
            };

// On update we want to check that we have this device and then update it
            watcher.Updated += async (DeviceWatcher sender, DeviceInformationUpdate deviceInfoUpdate) =>
            {
                BluetoothDevice dev = null;

                // lock the list while we operate on it
                lock (devices)
                {
                    // Find the device in the list
                    dev = devices.FirstOrDefault(d => d.Id.Equals(deviceInfoUpdate.Id));
                }

                if (dev != null)
                {
                    DeviceInformation devInfo = await DeviceInformation.CreateFromIdAsync(dev.Id);
                    // Update the device
                    devInfo?.Update(deviceInfoUpdate);

                    // Get the ble device for this id
                    BluetoothDevice updatedBleDevice = await BluetoothDevice.FromIdAsync(devInfo.Id);

                    // lock the list while we operate on it
                    lock (devices)
                    {
                        dev = updatedBleDevice;
                    }
                }
            };
```

In the provided snippets, I am able to convert from a `Windows.Devices.Enumeration.DeviceInformation` to a `InThehand.Bluetooth.BluetoothDevice`, but I am not able to reverse this action by using `DeviceInformation devInfo = await DeviceInformation.CreateFromIdAsync(dev.Id);` as the `BluetoothDevice.Id` is no longer the Id that created it.

```
Eg:
BluetoothDevice.Id = "68E68594A7B8"
DeviceInformation.Id = "BluetoothLE#BluetoothLEe8:48:b8:c8:20:00-68:e6:85:94:a7:b8"
```

The exception I receive from this is

```csharp
System.IO.FileNotFoundException
  HResult=0x80070002
  Message=The system cannot find the file specified. (Exception from HRESULT: 0x80070002)
  Source=mscorlib
  StackTrace:
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter`1.GetResult()
   at consoleTest.Program.<Main>d__0.MoveNext() in Program.cs:line 32

```

Am I missing something in how to convert back and forth? Is there a solution to this? I was able to retrieve the RSSI and the IsPresent properties from DeviceInformation, but not from the BluetoothDevice. I am trying to expose these properties and thought I could just convert back and forth to retrieve them using the `"System.Devices.Aep.SignalStrength"` property for example.

peterfoot, 2023-05-03:
There is an implicit cast between BluetoothDevice and the native Windows BluetoothLEDevice. You can do something like this

var btdev = (BluetoothLEDevice)device;
var di = btdev.DeviceInformation;

It's probably worth me adding an implicit operation to the DeviceInformation directly for interop with other APIs.

ThePapaG, 2023-05-04:
Thank you @peterfoot , This implicit cast does what I need. I was able to expose RSSI through this method
```csharp
public int RSSI
        {
            get
            {
                var bleDev = (Windows.Devices.Bluetooth.BluetoothLEDevice)_device;
                DeviceInformation devInfo = Task.Run(() => DeviceInformation.CreateFromIdAsync(bleDev.DeviceInformation.Id, new string[] { "System.Devices.Aep.SignalStrength" }).AsTask()).Result;
                return (int)devInfo.Properties["System.Devices.Aep.SignalStrength"];
            }
        }
```

closed 2023-05-04 by ThePapaG