inthehand/32feet #298: GetCharacteristicAsync returning null for Guuid created BluetoothUuid

ThePapaG, 2023-05-03:
Hi,

I am developing for windows. I am able to use short Ids to get characteristics and it works as so

```csharp
var devInfoService = Task.Run(() => _device.Gatt.GetPrimaryServiceAsync(BluetoothUuid.FromShortId(Convert.ToUInt16("180A", 16)))).Result);
var fwStringCharacteristic = Task.Run(() => devInfoService.GetCharacteristicAsync(devInfoService , BluetoothUuid.FromShortId(Convert.ToUInt16("2A26", 16)))).Result;
```

However, I am trying to interact with a custom Service and Characteristic, of which the characteristic returns null. I have been able to discover all services, characteristics, and descriptors on the device. (Note that bleDevice is a wrapper I have written. Where necessary I have written in the 32Feet calls)

```csharp
bleDevice.Connect();
                    var services = bleDevice.DiscoverServices();

                    foreach (var service in services)
                    {
                        Console.WriteLine($"Service {service.Uuid}");
                        var characteristics = bleDevice.DiscoverCharacteristics(service);
                        Console.WriteLine("\tCharacteristics:");
                        foreach (var characteristic in characteristics)
                        {
                            Console.WriteLine($"\t\t{characteristic.Uuid}");

                            var descriptors = bleDevice.DiscoverDescriptor(characteristic);
                            foreach (var descriptor in descriptors)
                            {
                                Console.WriteLine($"\t\t\t{descriptor.Uuid}");
                            }
                        }
                    }

Service 1800
        Characteristics:
                2A00
                2A01
                2A04
                2AA6
Service 1801
        Characteristics:
                2A05
                        2902
Service 98ac2a71-f935-4cef-a204-e52942478e08
        Characteristics:
                98ac2a72-f935-4cef-a204-e52942478e08
                98ac2a77-f935-4cef-a204-e52942478e08
                98ac2a78-f935-4cef-a204-e52942478e08
                98ac2a75-f935-4cef-a204-e52942478e08
                98ac2a73-f935-4cef-a204-e52942478e08
                98ac2a74-f935-4cef-a204-e52942478e08
                98ac2a76-f935-4cef-a204-e52942478e08
                98ac2a7a-f935-4cef-a204-e52942478e08
                        2902
Service 4e1dec02-9fb5-11ea-bb37-0242ac130002
        Characteristics:
                4e1dec03-9fb5-11ea-bb37-0242ac130002
                        2902
                4e1dec04-9fb5-11ea-bb37-0242ac130002
                        2902
Service 6f9fdc14-619c-438f-b15e-09f3af61196e
        Characteristics:
                6f9fdc15-619c-438f-b15e-09f3af61196e
                6f9fdc16-619c-438f-b15e-09f3af61196e
                        2902
Service 180A
        Characteristics:
                2A29
                2A24
                2A25
                2A27
                2A26
```

I am trying to use the same methodology to read interract with `4e1dec03-9fb5-11ea-bb37-0242ac130002`, and so I wrote

```csharp
var service = Task.Run(() => _device.Gatt.GetPrimaryServiceAsync(new Guid("4e1dec02-9fb5-11ea-bb37-0242ac130002"))).Result);
var characteristic = Task.Run(() => service.GetCharacteristicAsync(service, BluetoothUuid.FromGuid(new Guid("4e1dec03-9fb5-11ea-bb37-0242ac130002")))).Result;
```

However, characteristic always returns null. Despite the discovery showing that it does infact exist.

ThePapaG, 2023-05-04:
I have looked into this further. I tried making my `GetCharacteristic` method perform a discovery on the service and then iterate and return the match.

This fails with a List length of 0 characteristics. However, as per my original comment, I am able to discover all services and characteristics on the service using this same method. It is almost like, after you discover characteristics on a service once, the service then has no characteristics again?

```csharp
/// <summary>
        /// Get the specified characteristic of this service for this device
        /// </summary>
        /// <param name="service">The service to request the characteristic from</param>
        /// <param name="UUID">The characteristic UUID to request</param>
        /// <returns>The charactersitic if present, null if error</returns>
        /// <exception cref="InvalidOperationException">Throws if this device is not connected</exception>
        /// <exception cref="ArgumentNullException">Throws if service is null.</exception>
        public GattCharacteristic GetCharacteristic(GattService service, BluetoothUuid UUID)
        {
            if (!isConnected) throw new InvalidOperationException("Device is not connected");
            if (service == null) throw new ArgumentNullException(nameof(service));
            //return Task.Run(() => service.GetCharacteristicAsync(UUID)).Result;

            var characteristics = DiscoverCharacteristics(service);

            foreach (var characteristic in characteristics)
            {
                if (characteristic.Uuid.ToString() == UUID.ToString()) return characteristic;
            }

            return null;
        }
```

This always returns null as the list is length 0. Now if I use my first loop where i am writing out all of the characteristics, and store a local reference to the characteristic I care about from that list, then it works fine.

I have tested doing this print all twice, once after the other and the characteristics for the custom long UUIDs does not display the second time.

```
Service 1800
        Characteristics:
                2A00
                2A01
                2A04
                2AA6
Service 1801
        Characteristics:
                2A05
                        2902
Service 98ac2a71-f935-4cef-a204-e52942478e08
        Characteristics:
                98ac2a72-f935-4cef-a204-e52942478e08
                98ac2a77-f935-4cef-a204-e52942478e08
                98ac2a78-f935-4cef-a204-e52942478e08
                98ac2a75-f935-4cef-a204-e52942478e08
                98ac2a74-f935-4cef-a204-e52942478e08
                98ac2a76-f935-4cef-a204-e52942478e08
                98ac2a7a-f935-4cef-a204-e52942478e08
                        2902
Service 4e1dec02-9fb5-11ea-bb37-0242ac130002
        Characteristics:
                4e1dec03-9fb5-11ea-bb37-0242ac130002
                        2902
Service 6f9fdc14-619c-438f-b15e-09f3af61196e
        Characteristics:
                6f9fdc15-619c-438f-b15e-09f3af61196e
                6f9fdc16-619c-438f-b15e-09f3af61196e
                        2902
Service 180A
        Characteristics:
                2A29
                2A24
                2A25
                2A27
                2A26
Service 1800
        Characteristics:
                2A00
                2A01
                2A04
                2AA6
Service 1801
        Characteristics:
                2A05
                        2902
Service 98ac2a71-f935-4cef-a204-e52942478e08
        Characteristics:
Service 4e1dec02-9fb5-11ea-bb37-0242ac130002
        Characteristics:
Service 6f9fdc14-619c-438f-b15e-09f3af61196e
        Characteristics:
Service 180A
        Characteristics:
                2A29
                2A24
                2A25
                2A27
                2A26
```

923887863, 2023-05-29:
Hi, I have the same problem in my Console App.
It is work ok, but when I restart my BLE device, I call `GetCharacteristicAsync` and `GetCharacteristicsAsync` returns null.
Should I do something for reconnect?

ThePapaG, 2023-05-29:
My approach to my wrapper was that on my first get, I discover all characteristics and services and store them in a dictionary. Then on future gets I retrieve them from the dictionary.