inthehand/32feet #300: RemoteGattServer.ConnectAsync throwing object disposed exception

ThePapaG, 2023-05-08:
Hi,

I am trying to connect to the gatt server so that I can perform some operations on it. I am running x64 windows10.

```csharp
/// <summary>
        /// Connect to the device
        /// </summary>
        public void Connect()
        {
            if (!isConnected) BluetoothDevice.Gatt.ConnectAsync().Wait();
        }
```

This was working well until all of a sudden, I get an exception thrown non stop. Now it seems I am unable to connect to any of my endpoint devices (not just the one I was performing the testing on). The device is present, as I have a scanner that is scanning for devices that are present, and then I am selecting this device from that list.

```
System.AggregateException
  HResult=0x80131500
  Message=One or more errors occurred.
  Source=mscorlib
  StackTrace:
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task.Wait(Int32 millisecondsTimeout, CancellationToken cancellationToken)
   at System.Threading.Tasks.Task.Wait()
   at zencontrol.Transport.Bluetooth.BleDevice.Connect() in C:\Users\grant-dare\Documents\git_projects\dgevolt_git\zencontrol.Transport.Bluetooth\zencontrol.Transport.Bluetooth\BleDevice.cs:line 62
   at consoleTest.Program.<Main>d__1.MoveNext() in C:\Users\grant-dare\Documents\git_projects\dgevolt_git\zencontrol.Transport.Bluetooth\zencontrol.Transport.Bluetooth.Tools.Console\Program.cs:line 48

  This exception was originally thrown at this call stack:
    [External Code]

Inner Exception 1:
ObjectDisposedException: The object has been closed. (Exception from HRESULT: 0x80000013)
```

I don't understand the error, why it has started happening, how to fix it, how to handle it programmatically?

ThePapaG, 2023-05-08:
I am not using the BluetoothClient because it was not clear to me how I would write to a characteristic and receive a notification. My devices use this method as a way of sending responses.

Using the RemoteGattServer I am able to enable notifications on the characteristic and then write to it.

I am starting to wonder if the socket code inside BluetoothClient might in some way fix the error I am experiencing? However I have just attempted to connect in this way and instead get

```
System.Net.Sockets.SocketException
  HResult=0x80004005
  Message=A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond EE58A84F34DD:4e1dec03-9fb5-11ea-bb37-0242ac130002 (0)
  Source=System
  StackTrace:
   at System.Net.Sockets.Socket.InternalEndConnect(IAsyncResult asyncResult)
   at System.Net.Sockets.Socket.EndConnect(IAsyncResult asyncResult)
   at System.Threading.Tasks.TaskFactory`1.FromAsyncCoreLogic(IAsyncResult iar, Func`2 endFunction, Action`1 endAction, Task`1 promise, Boolean requiresSynchronization)
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.GetResult()
   at consoleTest.Program.<Main>d__1.MoveNext() in Program.cs:line 62

```

peterfoot, 2023-05-08:
BluetoothClient is for Bluetooth Classic (RFComm) connections and can not be used for Bluetooth LE. Do you keep a reference to the RemoteGattServer in your containing class for the lifetime of the connection?

ThePapaG, 2023-05-08:
Hi Peter, 

My wrapper class, `BleDevice` contains a public member `BluetoothDevice` which is of type `InTheHand.Bluetooth.BluetoohDevice` of which I access the `RemoteGattServer` through. At the time of calling I have checked and my wrapper, BluetoothDevice, and Gatt are all non null and contain values.

ThePapaG, 2023-05-08:
I have now tried copying the [platform connect method](https://github.com/inthehand/32feet/blob/main/InTheHand.BluetoothLE/Platforms/Windows/RemoteGattServer.windows.cs#L34) directly into my project and casting the `IntTheHand.Bluetooth.BluetoothDevice` to `Windows.Devices.Bluetooth.BluetoothLEDevice` to get the `NativeDevice` and the same error is thrown on `RequestAccessAsync`

ThePapaG, 2023-05-08:
I actually think I see the problem now in my setup. My wrapper is disposable, which I am using in a using block. On dispose, I disconnect.

I now see that the `BluetoothDevice.Disconnect` calls `PlatformCleanup` which will dispose of native objects. 
https://github.com/inthehand/32feet/blob/main/InTheHand.BluetoothLE/Platforms/Windows/RemoteGattServer.windows.cs#L88

I am not sure I agree with the decision that disconnect also cleans up native objects. What if I want to connect and disconnect multiple times during the life of the object?

rather, I would suggest that the class be `IDisposable` and therefore cleanup on dispose/finalise.

The use case that was all of a sudden causing this error was that, I can get the RSSI through native windows. So my scanned devices I was printing out, then constructing my wrapper which gets the RSSI from the native windows object, then disposing and disconnecting so that I don't leave derelict connections. The `BluetoothDevice` object never left scope, but the `BleDevice` wrapper was. Therefore it was disposing and disconnecting, and in turn disposing the native object and throwing this error.

ThePapaG, 2023-05-08:
The connection isn't performed on construction, but the destruction is performed on disconnection. The concepts of are not linked at both ends which is what caused the issue for me

peterfoot, 2023-05-09:
The problem with the Windows API is that there is no explicit disconnect method - you have to dispose the native object to ensure it disconnects. To allow multiple disconnections and reconnections requires the library to smartly recreate new native objects.