inthehand/32feet #304: Bluetooth file transfer Windows to Android

DavidR5, 2023-05-23:
Firstly, thank you for the work you have put into this project, great job.
I am working on a C# .Net 7 Maui Blazor app that needs to transfer files from Windows and Android. Currently testing on Windows 10, Android 9, 10 and 11 using InTheHand libraries . 
Android to Android works fine and transfer using Windows built in Bluetooth file transfer works fine.
Windows to Android works fine for a 30 kb but fails for an 83 kb file with "RequestedUrlTooLarge|Final"
This error appears to be thrown by ObexMethod.Put in ObexWebRequest.cs->AddBody.
I have included my code below.
Any help with resolving this issue would be much appreciated.
Thanks
David
    private async void sendFile()
    {
        ObexWebResponse response = null;
        try
        {
            var result = await FilePicker.PickAsync();
            if (result != null)
            {

                string strDeviceAddress = selectedDevice.DeviceAddress;
                
                System.Uri uri = new Uri("obex://" + strDeviceAddress + "/" + System.IO.Path.GetFileName(result.FileName));
                ObexWebRequest request = new ObexWebRequest(uri);
                request.ReadFile(result.FullPath);
                response = (ObexWebResponse)request.GetResponse();
                progressMessage = response.StatusCode.ToString() ;
                response.Close();
            }
        }
        catch (Exception ex)
        {
            progressMessage = ex.Message;
            response.Close();
        }
    }

DavidR5, 2023-06-20:
Hi Peter,
In ObexParser.cs, line 64 sets remoteMazPacket to 0 causing line 66 to set remoteMaxPacket = ushort.MaxValue; which is 65535 which causes the error.
                        remoteMaxPacket = (ushort)IPAddress.NetworkToHostOrder(BitConverter.ToUInt16(packet, pos + 2));
                        if (remoteMaxPacket == 0)
                            remoteMaxPacket = ushort.MaxValue;
If I set remoteMaxPacket to 0x800 here the file transfer works fine.
                        remoteMaxPacket = (ushort)IPAddress.NetworkToHostOrder(BitConverter.ToUInt16(packet, pos + 2));
                        if (remoteMaxPacket == 0)
                            //remoteMaxPacket = ushort.MaxValue;
                            remoteMaxPacket = 0x800;
I will continue to look at why line 64 is not working and let you know if I can work it out. This is new to me so I am learning as I go.
David

DavidR5, 2023-06-20:
remoteMaxPacket = (ushort)(packet[pos + 2] << 8 + packet[pos + 3]); // works remoteMaxPacket 8192
Works for me from Windows to Android but will this break Windows to Windows which worked previously. Don't have a second Windows computer with me at present to test this.

DavidR5, 2023-06-20:
Does it need something like:
                        if (BitConverter.IsLittleEndian)
                        {
                            remoteMaxPacket = (ushort)(packet[pos + 2] << 8 + packet[pos + 3]); // works remoteMaxPacket 8192
                        }
                        else
                        {
                            remoteMaxPacket = (ushort)IPAddress.NetworkToHostOrder(BitConverter.ToUInt16(packet, pos + 2));
                        }

DavidR5, 2023-06-22:
Hi Peter,
So far I have tested the above changes with:
Windows 11 to Windows 11
Android 9 to Windows 11
Android 9 to Android 10
It worked fine in each of these scenarios.
Cheers,
David

peterfoot, 2023-06-24:
Thanks for taking the time to investigate this and provide detailed feedback. This looks like a good solution to the problem. Would you like to submit a pull request?

DavidR5, 2023-06-25:
Hi Peter,
I will get one of my colleagues to test on MAC first.
David

commit bac438d3 (2023-10-04): Obex RemoteMaxPacket fix #304
closed 2023-10-04 by peterfoot