inthehand/32feet #305: Application crashing on release "The type initializer for 'InTheHand.Android Activity' threw an exception."

GeraldoJansen, 2023-05-25:
Hey guys.

I'm using the **InTheHand.Net.Bluetooth** library for the **Android platform**, using **Xamarin Forms, C# language**, in **android version 13**, configured from **minimum version 21 to API version 33**, I prepared the environment, made a **standard project** and made the communication with bluetooth there and **referenced it in the Xamarin Forms project**. In the **debug** the tests were successful, now when I put it in the **release** the app breaks and throws an error "**One or more errors occurred. (The type initializer for 'InTheHand.AndroidActivity' threw an exception.)**" as android in the recent version is requesting more explicit permissions for "BluetoothScan" and "BluetoothConnect" **does not get back versions of Xamarin.Essentials and Xamarin.Forms**.

I thought it could be down to the structure and I jogged all the communication to the **Xamarin Forms project and it continues with the same result**. And I even put the following line in the MainActivity and it didn't work: **_"InTheHand.AndroidActivity.CurrentActivity = this;"_**

> Note: It does not point out any code errors, only the following warnings:
1. Severity Code Description Project File Line Suppression Status
Warning NU1605 Package downgrade detected: Xamarin.Essentials from 1.7.1 to 1.7.0. Reference the package directly from the project to select a different version.
 TesteImpressao.Android -> TesteImpressao -> Xamarin.Essentials (>= 1.7.1)
 TesteImpressao.Android -> Xamarin.Essentials (>= 1.7.0) 1

2. Warning NU1605 Package downgrade detected: Xamarin.Forms from 5.0.0.2337 to 5.0.0.2196. Reference the package directly from the project to select a different version.
 TesteImpressao.Android -> TesteImpressao -> Xamarin.Forms (>= 5.0.0.2337)
 TesteImpressao.Android -> Xamarin.Forms (>= 5.0.0.2196) 1

I need a new version of my project urgently. How can we solve this problem?

GeraldoJansen, 2023-05-25:
@peterfoot I changed the Xamarin.Essentials version to 1.7.0 and the app keeps crashing on Release.
Detail: See if this information helps you. When I mark this option the application works in the release:
[image]

I'll prepare an example so you can better help me.

GeraldoJansen, 2023-05-25:
Here is a project with an example of how my project is consuming InTheHand.Net.Bluetooh:
https://github.com/GeraldoJansen/TesteImpressaoInTheHand
Thank you for your attention. I await your help.

> Note: Remembering that I already consumed the nuget with the direct consumption classes in the main Xamarin Forms project.

peterfoot, 2023-05-25:
Thanks for providing code to reproduce this issue. I've done some testing and can see the error. Working on a fix for InTheHand.AndroidActivity.

GeraldoJansen, 2023-05-25:
Awesome! Glad the sample project helped. Waiting for your correction then.

peterfoot, 2023-05-25:
Still working on the issue, but this should work as a workaround:-
in your Android Activity OnCreate method add the following after Xamarin.Essentials.Platform.Init(this, savedInstanceState);

var a = Xamarin.Essentials.Platform.CurrentActivity;

Then the following calls to CurrentActivity should work as expected. Make sure you don't get or set AndroidActivity.CurrentActivity before the call to Platform.Init

GeraldoJansen, 2023-05-26:
I did some tests here @peterfoot. It worked as expected in release mode. Thank you very much.
It looked like this:

[image]

I will wait for the final solution.

GeraldoJansen, 2023-05-26:
@peterfoot clarify a doubt for me. Do you have any resource in your project, InTheHand.Net.Bluetooth to check if the cell phone has Bluetooth and if Bluetooth is turned on? I'm using an interface and a class in my external dll to be able to have this feature of "Xamarin.Essentials" as follows:

[image]

If yes, can you give me an example?

peterfoot, 2023-05-29:
You can use BluetoothRadio.Mode to determine the current mode of the BluetoothAdapter. However this doesn't work on iOS as the API there is much more restrictive.