inthehand/32feet #318: MAUI iOS - Bluetooth.ScanForDevicesAsync does not return devices, but RequestDeviceAsync does

rutgerkra, 2023-07-01:
I'm trying to communicate with BLE devices in my MAUI app. To provide a better UX I want to filter the scanned BLE devices and show them in my own UI. However, I can't seem to make `Bluetooth.ScanForDevicesAsync` return anything other than an empty collection.

Here's how I'm using it:

```csharp
var devices = await Bluetooth.ScanForDevicesAsync(new RequestDeviceOptions() { AcceptAllDevices = true, });

foreach (var device in devices)
{
    AvailableDevices.Add(device.Name);
}
```

When I use `var selectedDevice = await Bluetooth.RequestDeviceAsync(new RequestDeviceOptions() { AcceptAllDevices = true });`, the popup that shows up does give devices and I'm also able to communicate with them.

Am I using `Bluetooth.ScanForDevicesAsync` wrong? Or is this not working properly?

Running on iPadOS 16.5.1.

peterfoot, 2023-07-06:
ScanForDevices (programmatic scan of unpaired devices) is not currently implemented for iOS. You can access paired devices or use the RequestDevice picker.

rutgerkra, 2023-07-06:
Good to know, thanks!

peterfoot, 2023-07-06:
I'm going to reopen this to track progress on adding this functionality.

peterfoot, 2024-04-16:
The ScanForDevices method has not been implemented on iOS yet. There is no specific reason why it cannot - it'll be similar to the implementation already used to build the device picker for RequestDeviceAsync but I have not had the time to complete it.

acaliaro, 2024-04-30:
Hi @peterfoot , thanks for the great work! I use your libraries since wince.... do you have a possible date for this request?