inthehand/32feet #319: Mtu return 0 after request

vitalii-vov, 2023-07-05:
Hi,
When I request the MTU using the RequestMtu method, I always get 0 in Gatt.Mtu property.

```C#
var res = _device.Gatt.RequestMtu(250);
var mtu = _device.Gatt.Mtu;
Debug.WriteLine($"mtu is: {res} res: {mtu}");
```
Console log:
```
[0:] mtu is: True res: 0
```

But, if I use Delay, MTU will change
```C#
var res = _device.Gatt.RequestMtu(250);
await Task.Delay(200);
var mtu = _device.Gatt.Mtu;
Debug.WriteLine($"mtu is: {res} res: {mtu}");
```
Console log:
```
[0:] mtu is: True res: 250
```

Perhaps you should make the RequestMtu method asynchronous so that after calling it you can be guaranteed to get the MTU value on the device?

peterfoot, 2023-07-06:
what platform is this on?

vitalii-vov, 2023-07-06:
On Android

peterfoot, 2023-08-11:
Reworked in 4.0.34 as RequestMtuAsync - returns on success or failure (inc timeout if no response).

vitalii-vov, 2023-08-22:
@peterfoot thanks for solving the problem
~~~C#
var res = await _device.Gatt.RequestMtuAsync (250);
var mtu = _device.Gatt.Mtu;
Debug.WriteLine($"mtu is: {res} res: {mtu}");
~~~
Printed
~~~
mtu is: false res: 250
~~~
I do not understand, if `res == false` then we are not sure that `mtu == 250`, correct?

peterfoot, 2023-08-22:
I think I can see the bug here. When the OnMtuChanged callback is called although it returns the status and this is used to return true or false it also sets the RemoteGattServer.Mtu to the value. This callback is actually saying it failed to set the Mtu to 250 and so the value should be discarded. I will fix this for the next release. For now if the RequestMtuAsync returns false you cannot rely on the Mtu value. You can try the request again with a different value and see if that works.

commit 4b1c6370 (2023-08-22): fix: #319 Only store Mtu if change was successful.
closed 2023-08-22 by peterfoot