inthehand/32feet #323: How do I get a list of already connected BLE devices on my Win10?

Ugobyte, 2023-07-06:
Hello,

I am using VS2022, my application is .NET6.0 and I want to get a list of all the BLE devices that are already connected to my Windows 10 PC. These BLE devices are mice, keyboard and Gamepads.

My code looks like this:
[image]

However, it never finds any BLE device. When I debug my code, I get this:
[image]

What am I doing wrong? Can someone help point me in the right direction? Any code samples will be greatly appreciated.

Ugobyte, 2023-07-06:
I was able to get it to work. However, the search only returns my neighbors Samsung TV, hahaha.
I want the search to include BLE devices that are ALREADY connected to my Win10 PC.

The image below shows I have a Bluetooth Touch Mouse and a Bluetooth Gamepad already connected, how can I find the Touch Mouse and Gamepad using inthehand? My goal is to make a Windows app that allows me communicate with an ALREADY connected/paired device.
[image]

peterfoot, 2023-07-06:
BluetoothClient is part of the Bluetooth Classic API. The separate InTheHand.BluetoothLE contains Bluetooth LE functionality.

Ugobyte, 2023-07-07:
> InTheHand.BluetoothLE

Thanks Peter.
I've gone ahead an downloaded the inthehand.BluetoothLE package.
Can you point me where I can see example code or some documentation? I've looked all over but I can't find a thing.

peterfoot, 2023-07-07:
There is a very rough sample here - [Samples/BluetoothClientApp](https://github.com/inthehand/32feet/tree/main/Samples/BluetoothClientApp)

The API documentation is here - https://inthehand.github.io/html/N_InTheHand_Bluetooth.htm

Ugobyte, 2023-07-07:
> There is a very rough sample here - [Samples/BluetoothClientApp](https://github.com/inthehand/32feet/tree/main/Samples/BluetoothClientApp)
> 
> The API documentation is here - https://inthehand.github.io/html/N_InTheHand_Bluetooth.htm

Pardon me for asking such questions.
But when I call  `Bluetooth.ScanForDevicesAsync()` i get a **System.PlatformNotSupportedException: 'Operation is not supported on this platform.'** 

I'm on a Win10 PC, using VS2022 with .NET6.0 and I'm trying to make a ClassLibrary project (I'm writing a custom .dll)
Am i getting this exception due to the fact that i writing a .dll? and not a WPF or Windows form project?

Also, I see in the remarks where you guys mentioned **" Windows - Add the Bluetooth permission"** 
Perhaps I'm getting this exception because I did not add Bluetooth permission.
If so, how do i "add" Bluetooth permission?

Thanks Peter, you've been great thus far.

peterfoot, 2023-07-07:
It sounds like the wrong DLL is being deployed. Does the consuming app target net6.0-windows10.x or higher?

Ugobyte, 2023-07-08:
> 

Yes.
[image]

Ugobyte, 2023-07-08:
[image]

Ugobyte, 2023-07-10:
I have the latest InTheHand.BluetoothLE which is version 4.0.33. And my target platform is net-3.0-windows10.017763.0. 
Are these incompatible ?

Ugobyte, 2023-07-12:
@peterfoot Do you have any ideas? or suggestions?

berkut0, 2023-07-26:
@Ugobyte confirm
Same error when trying to call ScanForDevicesAsync with BLE

Ugobyte, 2023-07-28:
@berkut0 I will try it shortly and report back.

Ugobyte, 2023-08-17:
> @Ugobyte confirm Same error when trying to call ScanForDevicesAsync with BLE

Yes i can confirm I'm using .ScanForDevicesAsync. Here is my code 
[image]
.
.
.
.
.
When I run, I get this error.
[image]