inthehand/32feet #324: Problems with my deployment of the project to hololens2:DllNotFoundException: Unable to load DLL 'ws2_32.dll'.

ilves9587, 2023-07-07:
The "InTheHand.Net.Bluetooth" can perform good in the Unity 3D 
but after deployment of the project to the Hololens2 there is bug shown .

DllNotFoundException: Unable to load DLL 'ws2_32.dll'. Tried the load the following dynamic libraries: Unable to load dynamic library 'ws2_32.dll' because of 'C:\buildslave\unity\build\External\il2cpp\builds\external\baselib\Source\WinApi\Baselib_DynamicLibrary_WinApi.inl.h(36):Baselib_DynamicLibrary_OpenUtf16: Failed to open the requested dynamic library (0x06000000) - The specified module could not be found.  (WinError:0000007e)
  at InTheHand.Net.Sockets.Win32Socket+NativeMethods.socket (System.Net.Sockets.AddressFamily af, System.Net.Sockets.SocketType type, System.Net.Sockets.ProtocolType protocol) 

How can I solve this issue? 
Thank you so much for your reading.

peterfoot, 2023-07-07:
Yes, this is a complicated one because there are two Windows APIs for Bluetooth classic - the original Win32 using Sockets and the Windows 10 API. Hololens doesn't have the legacy Win32 APIs and that is why you are seeing this error.

At the moment there isn't a single dll build of InTheHand.Net.Bluetooth containing both implementations so it depends what your consuming project targets as to which dll you get from the NuGet package. A UWP App will always get the Windows 10 API. If you use Unity in .NET 4.x compatibility it will use the .NET 4.6.1 dll which uses the Win32 API. 

With Windows versions prior to 10 leaving support it might be the case that we eventually switch the .NET4.6.1 dll to use the Windows 10 API, but at the moment I know there are still people supporting Windows 7 and 8.1 and using the library.

AbhiG7, 2024-05-01:
I'm getting pretty much the exact same error but for a different dll. My error is

`DllNotFoundException: Unable to load DLL 'kernel32.dll'. Tried the load the following dynamic libraries: Unable to load dynamic library 'kernel32.dll' because of 'C:\build\output\unity\unity\External\il2cpp\builds\external\baselib\Source\WinApi\Baselib_DynamicLibrary_WinApi.inl.h(36):Baselib_DynamicLibrary_OpenUtf16: Failed to open the requested dynamic library (0x06000000) - The specified module could not be found.  (WinError:0000007e)`

For my project, I'm trying to communicate to the hololens from my laptop over tcp sockets. My program works locally but gives this error when I deploy to the hololens. How can I get the right packages or DLLs so that my program runs on the hololens?