inthehand/32feet #331: Error on search for bluetooth devices: failed to load assembly

olivertdam, 2023-07-28:
Hello,

I am trying to create an app for Android to connect to a bluetooth device. I am using InTheHand.Net, but get the following error message whenever I try to search for devices.

error message:
`[monodroid-assembly] open_from_bundles: failed to load assembly InTheHand.Net.Bluetooth.dll
Loaded assembly: /data/data/com.companyname.shuntapp2/files/.__override__/InTheHand.Net.Bluetooth.dll [External]
[monodroid-assembly] open_from_bundles: failed to load assembly InTheHand.AndroidActivity.dll
Loaded assembly: /data/data/com.companyname.shuntapp2/files/.__override__/InTheHand.AndroidActivity.dll [External]
[monodroid-assembly] open_from_bundles: failed to load assembly Uno.dll
[monodroid-assembly] open_from_bundles: failed to load assembly Uno.dll`

My code that gives me troubles is as:

```
public async void InitialBluetoothScan(object sender, EventArgs e)
        {     
            // new bluetooth client
            BluetoothClient client = new BluetoothClient();
            BluetoothDeviceInfo[] devices = (BluetoothDeviceInfo[])client.DiscoverDevices();

            foreach (var i in devices)
            {
                Console.WriteLine($"Name: {i.DeviceName}. Address: {i.DeviceAddress}");
            }
        }
```

I am trying to follow along this guide: https://medium.com/the-pragmatic-tech-review/how-to-transfer-data-via-bluetooth-in-net-a23a9d6a77f6

I have tried running the code on two separate computers and get the same error on both.

peterfoot, 2023-07-28:
How is linking configured in your project - select none or SDK Assemblies only to avoid the linker stripping out assemblies which are not referenced directly.

olivertdam, 2023-08-02:
Thank you for replying so quickly.
Based on peterfoot's suggestion i looked up linking and I think adding the following to my .csproj file in the `<ItemGroup>` section solved my problem:
```
<TrimmerRootAssembly Include="InTheHand.BluetoothLE" />
<TrimmerRootAssembly Include="InTheHand.Net.Bluetooth" />
```
However, it only lead me to another problem: Now i get the following error (though only after IntheHand has been correctly loaded):

```
Loaded assembly: InTheHand.AndroidActivity.dll [External]
[monodroid-assembly] open_from_bundles: failed to load assembly Uno.dll
[monodroid-assembly] open_from_bundles: failed to load assembly Uno.dll
```
I suspect this is not directly related to IntheHand, so I will probably post another issue somewhere else.

olivertdam, 2023-08-02:
I just found this [post](https://inthehand.com/2023/02/10/dot-net-dlls-droid-dependencies/#comments) by you, @peterfoot , commenting on the issue I am now facing with the Uno.dll assembly. 

You end the post by saying that the nuget package InTheHand.BluetoothLE no longer depends on either xamarin or maui (and Uno??). 
Would you recommend I switch to using the InTheHand.BluetoothLE package rather than InTheHand.Net? Or is that depended on which type of application I aim to make?

peterfoot, 2023-09-01:
On Android, whether you use InTheHand.BluetoothLE (Bluetooth LE) or InTheHand.Net.Bluetooth (Bluetooth Classic) you should add the following to your MainActivity to setup the activity reference which is required by some APIs particularly using pickers:-

InTheHand.AndroidActivity.CurrentActivity = this;

closed 2023-10-11 by peterfoot

intendchristophpr, 2025-02-06:
> On Android, whether you use InTheHand.BluetoothLE (Bluetooth LE) or InTheHand.Net.Bluetooth (Bluetooth Classic) you should add the following to your MainActivity to setup the activity reference [...]

Thank you!
This more or less hidden post helped me a lot.
(Isn't it documented yet?)