inthehand/32feet #333: Bluetooth.ScanForDevicesAsync() return a list of null objects

digitalfr4nk, 2023-08-05:
Hi,

I have been trying to get a list of BLE devices, and using a few of your examples it seems I am stuck in a couple of places:

1. Both Bluetooth.ScanForDevicesAsync() and Bluetooth.GetPairedDevicesAsync() return correct number of items but they're all null!
2. BluetoothDevice device = await Bluetooth.RequestDeviceAsync(options); after selecting a device from the poopup, it returns null. (again this is from the example.)

[image]

Any ideas? (I am using a UWP app, Win11, VS2022)

digitalfr4nk, 2023-08-05:
I got it figured out, and I'll leave this here in case it helps any one else: 
I neglected to include `<DeviceCapability Name="bluetooth"/>` in the Package.appxmanifest.

closed 2023-08-05 by digitalfr4nk

peterfoot, 2023-08-06:
Glad you discovered the issue. However this shouldn't have happened because the behaviour is not clear. I will look at changing it to throw a descriptive exception when it fails with missing permission.