inthehand/32feet #337: Unable to receive data on Android using BluetoothClient and BluetoothListener

Speedberg, 2023-08-20:
I've been unable to receive any data on my Android device, regardless if I use `BluetoothClient` or `BluetoothListener`. 

When using `BluetoothClient`, the Android device is able to successfully connect and send data to my PC (which is using `BluetoothListener`) - however, any data sent from my PC to the Android device is not received.

When using `BluetoothListener`, my PC (using `BluetoothClient`) is able to successfully connect to the Android device - however, the Android device again doesn't receive any data, and also `listener.Pending()` always returns `false`.

Android version: 13
Desktop OS: Windows
InTheHand.Net.Bluetooth: 4.0.36

Any help would be greatly appreciated :)

peterfoot, 2023-08-20:
Can you please share your code to replicate the issue?

Speedberg, 2023-08-20:
Here's the code for the client and the server:
```
using InTheHand.Net;
using System.Threading;
using System.Threading.Tasks;

namespace Pen.Net;

public sealed class BluetoothClient : BluetoothPeer, IClient
{
    public IConnection Connection => _connection;

    public bool IsConnected => _connection?.IsConnected ?? false;

    private BluetoothConnection _connection;

    public bool Connect(object address, object service)
    {
        _connection = new BluetoothConnection(this);
        var task = Task.Run(async () => await _connection.Connect((BluetoothAddress)address, Bluetooth.Service));
        task.Wait();
        return _connection.IsConnected;
    }

    public void Disconnect()
    {
        if (!IsConnected)
            return;
        _connection.Close();
    }

    public override void Poll()
    {
        //Receive new data from the server
        _connection.Receive();

        if(_connection.HasTimedOut)
        {
            Logging.Logger.Debug("Disconnecting!");
            OnDisconnected(_connection, DisconnectReason.TimedOut);
            Disconnect();
        }
    }

    public void ResetHearbeat()
    {
        _connection.ResetTimeOut();
    }
}
```

```
using System.Net;
using System.Net.Sockets;

using InTheHand.Net;
using InTheHand.Net.Bluetooth;
using InTheHand.Net.Sockets;

namespace Pen.Net;

public sealed class BluetoothServer : BluetoothPeer, IServer
{
    private bool _isRunning;
    private BluetoothListener _listener;
    private BluetoothConnection _connection;

    public bool IsRunning => _isRunning;

    public event EventHandler<IConnection> OnConnection;
[... 281 lines cut ...]
            case SocketError.Interrupted:
            case SocketError.NotSocket:
                _peer.OnDisconnected(this, DisconnectReason.TransportError);
                break;
            case SocketError.ConnectionReset:
            case SocketError.ConnectionAborted:
            case SocketError.Disconnecting:
                _peer.OnDisconnected(this, DisconnectReason.Disconnected);
                break;
            case SocketError.TimedOut:
                _peer.OnDisconnected(this, DisconnectReason.TimedOut);
                break;
            default:
                _peer.OnDisconnected(this, DisconnectReason.TransportError);
                break;
        }
    }

    public void ResetTimeOut()
    {
        _lastHearbeat = DateTime.UtcNow;
    }
}
```
The full project is quite large - I'm happy to create a test repo if needed. Please let me know if you need any additional information.

Speedberg, 2023-08-21:
I've created a minimal test repository which can be found [here](https://github.com/Speedberg/32feet-android-test-repo/tree/main).
There are comments in [this file](https://github.com/Speedberg/32feet-android-test-repo/blob/main/Bluetooth/ViewModels/MainViewModel.cs) which highlight where the issues occur.

peterfoot, 2023-09-05:
I can see a way to mitigate this in the source but currently if you call GetStream multiple times you'll get different instances of the wrapper AndroidNetworkStream this could cause issues with multiple reads/writes and also if the first goes out of scope and is collected it will dispose the underlying stream. Because the Android API doesn't use .NET Sockets the Client property returns null and this special class is used to return a NetworkStream to appear the same as the Sockets based implementations.

peterfoot, 2023-09-05:
also BluetoothListener.Pending always returns false on Android because there is no underlying API to determine if there is a pending connection. You have to just Accept and wait unfortunately

Speedberg, 2023-09-06:
I see, thank you for letting me know. Is there anything I can do to prevent this from happening? I've tried the following below, but am still unable to read data on my Android device.
```
using (Stream stream = client.GetStream())
{

    string message = "Hello, world!";
    byte[] buffer = new byte[100];

    while (_keepRunning)
    {
        if (client.Connected)
        {
            await stream.WriteAsync(System.Text.Encoding.UTF8.GetBytes(message));
            //Note: this hangs / blocks on Android
            int read = await stream.ReadAsync(buffer, 0, buffer.Length);

            if (read > 0)
            {
                Logger.Debug("Data read: {0}", read);
            }
        }
        await Task.Delay(10);
    }
}
```

peterfoot, 2023-09-06:
Have you tried calling Flush/FlushAsync after the WriteAsync?

Speedberg, 2023-09-06:
I tried your suggestion, but unfortunately the issue seems to persist. After calling `stream.ReadAsync`, the debug message afterwards is never written to the console.
```C#
using (Stream stream = client.GetStream())
{

    string message = "Hello, world!";
    byte[] buffer = new byte[100];

    while (_keepRunning)
    {
        if (client.Connected)
        {
            await stream.WriteAsync(System.Text.Encoding.UTF8.GetBytes(message));
            await stream.FlushAsync();
            Logger.Debug("Data written!");

            //Note: this hangs / blocks on Android
            int read = await stream.ReadAsync(buffer, 0, buffer.Length);
            await stream.FlushAsync();

            if (read > 0)
            {
                Logger.Debug("Data read: {0}", read);
            }

            Logger.Debug("Data read called!");
        }
        await Task.Delay(10);
    }
}
```

Speedberg, 2023-09-07:
Unfortunately this doesn't seem to work either:  `IsDataAvailable()` always returns `false`, and `Length` always returns `0`. My PC can read any data sent by the Android device, but the Android device still seems to not be receiving anything.
```C#
using (Stream stream = client.GetStream())
{

    string message = "Hello, world!";
    byte[] buffer = new byte[100];

    while (_keepRunning)
    {
        if (client.Connected)
        {
            await stream.WriteAsync(System.Text.Encoding.UTF8.GetBytes(message));
            await stream.FlushAsync();
            Logger.Debug("Data written!");
#if __ANDROID__
            Logger.Debug("Android!");
            if(stream.IsDataAvailable() || stream.Length > 0)
            {
                int read = await stream.ReadAsync(buffer, 0, buffer.Length);
                await stream.FlushAsync();

                if (read > 0)
                {
                    Logger.Debug("Data read: {0}", read);
                }
                Logger.Debug("Data read called!");
            }
#endif
        }
        await Task.Delay(10);
    }
}
```

peterfoot, 2023-09-07:
If the Length returns zero then the data hasn't been sent. Did you add a flush after the write on the PC side too? I think there is some difference here in how Windows buffers the stream versus Android and so adding Flush to both will ensure the data is sent.

Speedberg, 2023-09-07:
I've added the call to `Flush()` on the PC side, however the issue still seems to persist and `Length` still returns `0`.

PC
```C#
public async Task Host()
{
    if(_keepRunning)
        return;
    try
    {
        Logger.Debug("Starting listener...");
        BluetoothListener listener = new BluetoothListener(BluetoothService.SerialPort);
        BluetoothClient client = null;
        Stream stream = null;
        _keepRunning = true;
        listener.Start();

        string message = "Hello, world!";
        byte[] buffer = new byte[100];

        while(_keepRunning)
        {
            if(listener.Pending())
            {
                client = listener.AcceptBluetoothClient();
                stream = client.GetStream();
                Logger.Debug("Accepted client: {0}", client.RemoteMachineName);
            }

            if(client != null)
            {
                if(client.Connected)
                {
                    await stream.WriteAsync(System.Text.Encoding.UTF8.GetBytes(message));
                    await stream.FlushAsync();
                    
                    int read = await stream.ReadAsync(buffer,0,buffer.Length);
                    await stream.FlushAsync();
                    
                    if(read > 0)
                    {
                        Logger.Debug("Data read: {0}", read);
                    }
                }
            }
            await Task.Delay(10);
        }

        Logger.Debug("Stopping listener...");

        listener.Stop();
    } catch(System.Exception e)
    {
        Logger.Error("Server Error: {0}", e);
    }
}
```

Android
```C#
public async Task Client()
{
    if(_keepRunning)
        return;
    try
    {
        Logger.Debug("Picking device...");
        BluetoothDevicePicker picker = new BluetoothDevicePicker();
        var device = await picker.PickSingleDeviceAsync();
        if(device == null)
            return;
        Logger.Debug("Device chosen: {0}",device.DeviceAddress);

        BluetoothClient client = new BluetoothClient();
        Logger.Debug("Client attempting connection...");
        await client.ConnectAsync(device.DeviceAddress, BluetoothService.SerialPort);
        Logger.Debug("Client connection successful!");
        _keepRunning = true;

        using (Stream stream = client.GetStream())
        {

            string message = "Hello, world!";
            byte[] buffer = new byte[100];

            while (_keepRunning)
            {
                if (client.Connected)
                {
                    await stream.WriteAsync(System.Text.Encoding.UTF8.GetBytes(message));
                    await stream.FlushAsync();
                    Logger.Debug("Data written!");
#if __ANDROID__
                    Logger.Debug("Android: {0} {1}", stream.IsDataAvailable(), stream.Length);
                    if(stream.IsDataAvailable() || stream.Length > 0)
                    {
                        int read = await stream.ReadAsync(buffer, 0, buffer.Length);
                        await stream.FlushAsync();

                        if (read > 0)
                        {
                            Logger.Debug("Data read: {0}", read);
                        }
                        Logger.Debug("Data read called!");
                    }
#endif
                }
                await Task.Delay(10);
            }
        }
        Logger.Debug("Client disconnecting...");
        client.Dispose();
    } catch(System.Exception e)
    {
        Logger.Error("Client Error: {0}", e);
    }
}
```

peterfoot, 2023-09-08:
I'm working on a new sample to show the client and listener and the following works for me on both Windows and Android:-
```
BluetoothClient client = new BluetoothClient();
client.Connect(device.DeviceAddress, MyServiceUuid);
if(client.Connected)
{
    using (var str = client.GetStream())
    {
        StreamReader r = new StreamReader(str);
        var receivedString = r.ReadLine();
        ReceivedTextLabel.Text = receivedString;

        StreamWriter w = new StreamWriter(str);
        w.WriteLine("Hello Bluetooth server!");
        w.Flush();
        w.Close();
        r.Close();
    }
}
```

```
private async Task ListenerTask()
{
    var client = _listener.AcceptBluetoothClient();
    using(var stream = client.GetStream() )
    {
        var sw = new StreamWriter(stream);
        sw.WriteLine("Hello Bluetooth client!");
        sw.Flush();

        await Task.Delay(1000);

        var sr = new StreamReader(stream);
        var receivedString = sr.ReadToEnd();
        Dispatcher.Dispatch(() =>
        {
            ReceivedTextLabel.Text = receivedString;
        });
        

        sr.Close();
        sw.Close();
    }
}
```
I've used both StreamReader and StreamWriter to transfer text and always call Flush after writing. I'm building it into a more full featured sample a bit like the chat app from the original 32feet samples.

Speedberg, 2023-09-08:
Your code works on my devices too - it seems to be that there is an issue with directly reading bytes from the stream. I'm not quite sure why `StreamWriter` and `StreamReader` work but `stream.Read` doesn't.