inthehand/32feet #343: RequestDeviceAsync always return null on Android 13

MauriceChocoSwiss, 2023-09-07:
Hi, I've tried to use your library to test a Bluetooth app. I'm currently trying your Battery level sample on Android device (API 33, BluetoothLE nuget last version).
After Adding activity correctly and right permissions, it still not work. 

var device = await Bluetooth.RequestDeviceAsync(new RequestDeviceOptions { AcceptAllDevices = true });

This method always return null. With a near device or connected device it's the same. It return null on android. But it work on Windows 11 with the same devices.

I've tried the ScanForDevicesAsync method but even if i add Scan permission in manifest i got an user permission exception...

here is permissions : 
[image]

What's wrong ? 

Thank you for help !

MauriceChocoSwiss, 2023-09-07:
It's a permission issue, We don't prompt for the BLUETOOTH_CONNECT permission (proximity devices) and the RequestDeviceAsync() method doesn't prompt like said in method summary... Maybe it doesn't work on android 13

For the SCAN permission, by enabling it in app permission on Device, it work now !

peterfoot, 2023-09-07:
I've added #344 to track the issue where the library doesn't prompt for permissions when required.

peterfoot, 2023-10-23:
I've made it easier to prompt for permission in Xamarin/.NET MAUI projects with this add-on library:-
https://inthehand.com/2023/10/11/asking-for-permission/

MauriceChocoSwiss, 2023-11-21:
I've try your solution few days ago now and it work on all my platforms!

Thank you very much for your support and your work!!! 🙏

closed 2023-11-21 by MauriceChocoSwiss