inthehand/32feet #345: System.Runtime.InteropServices.InvalidComObjectException: "COM object that has been separated from its underlying RCW cannot be used."

vitalii-vov, 2023-09-08:
The application crashes when running in Release mode.

```
System.Runtime.InteropServices.InvalidComObjectException: "COM object that has been separated from its underlying RCW cannot be used."

 at System.Runtime.InteropServices.McgMarshal.GetInterface(__ComObject obj, RuntimeTypeHandle typeHnd) in f:\dd\ndp\fxcore\CoreRT\src\System.Private.Interop\src\Shared\McgMarshal.cs:line 1266
   at __Interop.ComCallHelpers.Call(__ComObject __this, RuntimeTypeHandle __typeHnd, Int32 __targetIndex)
   at __Interop.ForwardComStubs.Stub_11[TThis](__ComObject __this, Int32 __targetIndex)
   at System.IDisposable__Impl.Dispatcher.Dispose()
   at Windows.Media.Playback.MediaPlayer.Dispose() in D:\Projects\CompanyName\AppName\AppName.UWP\obj\x86\Release\ilc\intermediate\MCS.BLEUpdater.UWP.McgInterop\SafeTypes.g.cs:line 111160
   at InTheHand.Bluetooth.BluetoothDevice.DisposeAllNativeObjects()
   at InTheHand.Bluetooth.BluetoothDevice.Finalize()
```

peterfoot, 2023-09-08:
How is the BluetoothDevice got rid of - is this during application exit? Is it going out of scope before this?

vitalii-vov, 2023-09-08:
I have scanner class
```c#
public class BleScanner : IBleScanner
{
    private BluetoothLEScan _scan;

    public BleScanner()
    {
        Bluetooth.AdvertisementReceived += OnAdvertisementReceived;
    }

    public event EventHandler<BluetoothDevice> DeviceDiscovered;

    public async Task<bool> IsBluetoothIsTurningOn()
    {
        return await Bluetooth.GetAvailabilityAsync();
    }

    public async Task Start()
    {
        if (await IsBluetoothIsTurningOn())
            _scan = await Bluetooth.RequestLEScanAsync();
    }

    public void Stop()
    {
        _scan.Stop();
    }

    private void OnAdvertisementReceived(object sender, BluetoothAdvertisingEvent e)
    {
        if (e.Device is null)
            return;
        DeviceDiscovered?.Invoke(this, e.Device);
    }
}
```

I call Scan function on my ViewModel
```C#
internal partial class MainViewModel : ObservableObject 
{
   private readonly IBleScanner _bleScanner;
   public MainViewModel(INavigation navigation, IBleScanner bleScanner, IConnection ble, IUserDialogs dialogs, IUpdatesService updatesService, ICache cache)
   {
      _bleScanner = bleScanner;
      _bleScanner.DeviceDiscovered += OnDeviceDiscovered;
   }
  
   [RelayCommand] async void Appearing()
   {
      IsBluetoothTurningOn = await _bleScanner.IsBluetoothIsTurningOn();
      if (_firstStart)
      {
          _firstStart = false;
          await _bleScanner.Start();   // <- if I call this, app will randomly crush
      }
   }
}
```

vitalii-vov, 2023-09-28:
I have attached a minimal demo project in which you can reproduce the error.

To reproduce the error, you need to switch the project in Release to UWP from launch, the application will crash on its own after some time.
Sometimes you can click on a list item or uncomment the timer functions in the code to speed up the fall.

[Testing32feetCrush.zip](https://github.com/inthehand/32feet/files/12749587/Testing32feetCrush.zip)

peterfoot, 2023-09-28:
@vitalii-vov thanks, this is helpful. I will run some tests.

vitalii-vov, 2023-11-03:
@peterfoot 
It looks like https://github.com/inthehand/32feet/commit/9a402b820c92251312e6edd87c6a5fcbbc265222 is working.
At least in a test project.

vitalii-vov, 2023-11-28:
@peterfoot
Are you planning to release this update soon?

peterfoot, 2023-11-28:
Sorry, been busy. I've pushed the latest version as 4.0.35 to NuGet this morning