inthehand/32feet #349: Unable to send file of 100KB to a connected device over bluetooth

Additive-abc, 2023-09-22:
Hello Peter,

Hope you are doing good. I am mailing you regarding support required in 'InTheHand.net.Obex'(4.0.4) package.
i have been using this package to send file from my laptop to a connected bluetooth device.
below code explains it....
 using (var bluetoothClient = new BluetoothClient())
 {
     string filePath = @"D:\pdf_export\Patient.pdf";
     var fileName = Path.GetFileName(filePath);
     var address = bluetoothClient.PairedDevices.ToList()[1].DeviceAddress;  //For my testing purpose only
var uri = new Uri("obex://" + address + "/" + fileName);
ObexWebRequest request = new ObexWebRequest(uri);
request.ReadFile(filePath);
var result = request.GetResponse() as ObexWebResponse;
Console.WriteLine($"Result StatusCode: {result.StatusCode}");

Above code is executed on .net 6 environment. Above code always gives status code 'RequestUrlTooLarge' code always even if the file size is 83kb. when i try to send file of 40KB it works fine.

I have created another dummy application with .net framework dll 'InTheHand.Net.pernosal dll and was able to execute above code without any error.

Please help me out how to proceed with 'IntheHand.Net.Obex' package.

Thanks in advance.

Additive-abc, 2023-09-25:
@peterfoot Request you to please suggest a solution at the earliest. Thanks in advance.

peterfoot, 2023-09-28:
I can't provide a timeline to investigate and resolve this as it is a project that I do in my spare time unpaid. The issue is likely an error in the negotiation of maximum packet size between the sender and receiver. If you've noticed it occurring between particular device types that would be useful to try and debug and reproduce the issue.

DavidR5, 2023-10-03:
This sounds exactly like the issue #304 that I submitted. Still waiting for my colleague to test on Mac, my solution works on Android and Windows.

peterfoot, 2023-10-04:
I've pushed the change for #304 version 4.0.6 will be available on NuGet shortly.

Additive-abc, 2023-10-04:
@DavidR5 really thanks for your help. 
@peterfoot i appreciate your help but unfortunately the new version (4.0.6) you released unable to create a object Bluetooth client.
[image]

Additive-abc, 2023-10-04:
@peterfoot I have downgraded the version to 4.0.4 able to create BluetoothClient object successfully. 
[image]
Kindly suggest. Thanks in advance!

peterfoot, 2023-10-04:
Okay, this is a separate issue with the Bluetooth library if your consuming project uses a generic net6.0 target platform. This will be resolved in a separate update for InTheHand.Net.Bluetooth once that is refactored.

peterfoot, 2023-10-04:
Please try with 4.07 - this now targets the latest InTheHand.Net.Bluetooth release and that features changes to the platforms supported by the generic net6.0 and net7.0 target platforms.

Additive-abc, 2023-10-04:
@peterfoot Really thanks a lot its working fine now. I have done one round of smoke test. i will let you once i do a complete detailed testing. Thankyou very much.

peterfoot, 2023-10-11:
I'm going to close this now, but feel free to re-open if the issue is still present

DavidR5, 2023-10-11:
Hi Peter,
I am working on a parallel constructor to internal ObexListenerContext(Socket s) that takes a stream as the parameter, public ObexListenerContext(Stream s) which is nearly done, just not tested. Public so I can call it directly from my app code but could change ObexListener to call it as well. Unfortunately not getting a lot of time to work on it. Will let you know when I get it working and tested and send you the code.
Cheers,
David

DavidR5, 2023-10-23:
Hi Peter,
Sorry this could be a lengthy email, let me know if you prefer me to do a pull request but my changes need some explanation (I could add more comments to the file).
My aim has been to be able to run the same Maui Blazor app on both Windows and Android and be able to transfer files from the app on one device to be received by the app on the other device. I have been able to achieve this with the following changes but not as cleanly as I would like.

  *   ObexListenerContext.cs

                                                               i.      Added public ObexListenerContext(Stream rs) stream communication instead of socket for communication with Android

                                                             ii.      I made this public so I can call it directly instead of using ObexListener, I bypassed ObexListener and used BluetoothListener directly due to requiring a custom service id. It would be better to make ObexListener handle custom service id’s.

                                                           iii.      Added public string getPath and private string _path; to return the filename of the sent file as this information is not available from a Stream.

The other addition I required was to be able to do an Md5 check on the received file to make sure it is complete, to achieve this I:

  *   ObexHeader.cs
     *   Added ContentMd5 = 0x30, 0x30 to 0x3F as the range for user defined Obex headers
  *   ObexWebRequest:DoPut()
     *   Added section for ContentMd5
  *   ObexWebRequest:GetResponse()
     *   This is messy, I set the ContentMd5 header in my code when creating the new ObexWebRequest but found it was getting overwritten when you create the new headers in GetResponse(). I can see this would be necessary to handle multiple calls to GetResponse. Maybe pass the Md5 string or a WebHeaderCollection object as a parameter to GetResponse(), just thought of that and I like that approach better.
     *   Added:

                                                               i.      string contentMd5 = Headers["ContentMd5"]; // added DR

                                                             ii.      WebHeaderCollection responseHeaders = new WebHeaderCollection();

                                                           iii.       responseHeaders.Add("ContentMd5", contentMd5); // added DR

  *   ObexListenerContext:
     *   Added public string getSentMD5() to retrieve the original Md5 generated by the sender
I have attached the files I have modified. I have also attached BluetoothFileTransfer.razor which shows the methods I use to send and receive files.
We would really appreciate if we could get this functionality included in a future release.
Let me know if you have any specific tasks I can help with to expediate this process.
Regards,
David

David Rodgers
Scientific IT Manager
QAAFI - UQ
Ph: +61 435120055
Email: ***@***.******@***.***>
Websites: www.qaafi.uq.edu.au<https://urldefense.proofpoint.com/v2/url?u=http-3A__www.qaafi.uq.edu.au_&d=DwMFAg&c=tpTxelpKGw9ZbZ5Dlo0lybSxHDHIiYjksG4icXfalgk&r=pm59X2vpNJXrtyTMprsG-4AkKu6bu3fLkeqNL11haFw&m=ad_mqXCxf2lte2t_fQs8tpM6uT21hc9TmWONVuoSlOE&s=WcnQSfFK0EtyBC2QC0i2qs_dSXqd25OKYaSsqivgwRE&e=>
                www.ozsorghum.org.au<https://urldefense.proofpoint.com/v2/url?u=http-3A__www.ozsorghum.org.au_&d=DwMFAg&c=tpTxelpKGw9ZbZ5Dlo0lybSxHDHIiYjksG4icXfalgk&r=pm59X2vpNJXrtyTMprsG-4AkKu6bu3fLkeqNL11haFw&m=ad_mqXCxf2lte2t_fQs8tpM6uT21hc9TmWONVuoSlOE&s=COeg2Rzc5Z_YcYr0sub4S1itoszio1659VfFc4xcCH0&e=>

Hi Peter,
I am working on a parallel constructor to internal ObexListenerContext(Socket s) that takes a stream as the parameter, public ObexListenerContext(Stream s) which is nearly done, just not tested. Public so I can call it directly from my app code but could change ObexListener to call it as well. Unfortunately not getting a lot of time to work on it. Will let you know when I get it working and tested and send you the code.
Cheers,
David

// 32feet.NET - Personal Area Networking for .NET
//
// InTheHand.Net.ObexHeader
// 
// Copyright (c) 2003-2022 In The Hand Ltd, All rights reserved.
// This source code is licensed under the MIT License

using System;

namespace InTheHand.Net
{
    /// <summary>
    /// Header identifiers to describe some aspect of the object being transferred.
    /// </summary>
    public enum ObexHeader : byte
    {
        // OBEX Header codes
        None = 0x0,

        /// <summary>
        /// String describing the name of the object.
[... 1495 lines cut ...]
            do {
                readBytes = fs.Read(buffer, 0, buffer.Length);
                len += readBytes;
                requestStream.Write(buffer, 0, readBytes);
            } while (readBytes > 0);

            fs.Close();
            requestStream.Close();

            // write content length
            ContentLength = len;
        }
        #endregion

        #region Get Response
        /// <summary>
        /// Returns the OBEX server response.
        /// </summary>
        /// -
        /// <returns>An <see cref="T:InTheHand.Net.ObexWebResponse"/>.</returns>
        /// -
        /// <exception cref="System.Net.WebException">
        /// An error occurred, with the error that occured being stored in the 
        /// <see cref="P:System.Exception.InnerException"/> property.  If the error 
        /// occurred in the connect p

peterfoot, 2023-10-23:
Hi @DavidR5 

Thanks for putting the time into this! If you could put it into a pull request that would be great!
Thanks,

Peter

DavidR5, 2023-10-24:
Hi Peter,
I created a fork in my own account, added my changes to the fork and just want to check my settings before hitting the “Create pull request” button. I am nervous about having the base repository as your default repository. I don’t seem to be able to create a branch for myself within the inthehand/32feet repository.
Are these the correct settings?
Thanks,
David

***@***.***

peterfoot, 2023-10-25:
I've created a new branch "obex-development" so you can target this for your pull request.

DavidR5, 2023-10-25:
Thanks Peter,
This makes more sense.
I have created the pull request.
Regards,
David