inthehand/32feet #355: Upgrading to Windows Build >-20020

JongChern, 2023-10-23:
Hi, just checking if there are any plans to update to Windows Build >=20020. I am interested in testing out some of the new settings provided via https://learn.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.bluetoothlepreferredconnectionparameters?view=winrt-22000 

thanks

peterfoot, 2023-10-23:
Certainly always interested in moving with the new SDKs. Is there a particular setting you require?

JongChern, 2023-10-23:
in particular I am looking at this method [RequestPreferredConnectionParameters](https://learn.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.bluetoothledevice.requestpreferredconnectionparameters?view=winrt-22000) which allows you to set preferred properties such as [ThroughputOptimized](https://learn.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.bluetoothlepreferredconnectionparameters?view=winrt-22000). 

on a related note, do you by any chance know if there is a way to request for 2m phy from the windows side. I know it's [possible on android](https://github.com/inthehand/32feet/issues/175), but have not seen any options on windows.

peterfoot, 2023-10-23:
[GetConnectionPhy](https://learn.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.bluetoothledevice.getconnectionphy?view=winrt-22000) allows you to query it but there is no method to request an alternate PHY currently

JongChern, 2024-02-19:
hi @peterfoot was wondering if they are any plans for this. I am getting significantly better throughput when using Android so was hoping to try and figure out if there any any available methods on Windows which can help achieve similar results.