inthehand/32feet #359: BluetoothDeviceInfo from partial class to abstract

polipo, 2023-10-31:
In last commit of Bluetooth, there's change from partial class to abstract,
and I cannot resolve how to create an instance of BluetoothDeviceInfo
from macAddress.

Please, can help and explain how to do it? 

 deviceInfo = new BluetoothDeviceInfo(new InTheHand.Net.BluetoothAddress(ulong.Parse(Encoding.ASCII.GetString(_macAddress), System.Globalization.NumberStyles.HexNumber, System.Globalization.CultureInfo.InvariantCulture)));

Thanks

peterfoot, 2023-11-02:
Sorry, looks like this was a knock-on effect of changing the implementation to support multiple platform variants of BluetoothDeviceInfo. I think the solution is a single public class which then holds the platform specific instance internally, similar to how the BluetoothClient etc are implemented.

polipo, 2024-01-15:
Hi, have you a pratical solution or workaround for the last release 4.1.40?

Because i use to create a BluetoothClient by its macaddress like source above:

    _connection = new BluetoothClient();

    deviceInfo = new BluetoothDeviceInfo(new InTheHand.Net.BluetoothAddress(ulong.Parse(Encoding.ASCII.GetString(_macAddress), System.Globalization.NumberStyles.HexNumber, System.Globalization.CultureInfo.InvariantCulture)));

    if (deviceInfo == null)
    {
        { throw new ApplicationException("Device not paired"); }
    }
......

peterfoot, 2024-03-10:
Fixed in 4.1.42

closed 2024-03-10 by peterfoot

dilbie, 2024-03-14:
How was this fixed in 4.1.42? I am having the same issue and still can not find a public constructor for BluetoothDeviceInfo which accepts a BluetoothAddress.

peterfoot, 2024-03-14:
Sorry, a bit pre-emptive there. 42 contains the architectural change. 4.1.43 adds the address constructor.