inthehand/32feet #363: bluetoothClient.Connect(address, BluetoothService.Handsfree)) not working

abusaadp, 2023-11-13:
Hi,
I am trying to implement bluetooth dial in my iPhone from my Windows machine. I am using the following code 

`BluetoothAddress address;
address = BluetoothAddress.Parse("749EAF47B192");
using var bluetoothClient = new BluetoothClient();
await Task.Run(() => bluetoothClient.Connect(address, BluetoothService.Handsfree));
var phoneNumber = "9600020047";
using var stream = bluetoothClient.GetStream();
var cmds = new List<string>()
{
    "AT+CMER\r",
    "AT+CIND=?\r",
    "AT+BRSF=\r",
    $"ATD{phoneNumber};\r"
};
foreach (var cmd in cmds)
{
    Debug.WriteLine($"sending: {cmd}");
    var cmdData = Encoding.ASCII.GetBytes(cmd);
    await stream.WriteAsync(cmdData);
    await stream.FlushAsync();

    // could be coming from a pool, but for demo..
    var buffer = new byte[1024];
    var readResponseByteCount = await stream
        .ReadAsync(buffer, 0, buffer.Length);
    var responseData = buffer
        .Take(readResponseByteCount)
        .ToArray();
    var responseText = Encoding
        .ASCII
        .GetString(responseData)
        .Trim();
    // do something with the response
}`

But my code fails at the following line

`await Task.Run(() => bluetoothClient.Connect(address, BluetoothService.Handsfree));`

I get the following error:

`No connection could be made because the target machine actively refused it.` 

with error code 10061 

Please help me in fixing this issue. I also tried with the different iPhone device but I am getting the same issue.

peterfoot, 2023-11-13:
Connection will fail if the iPhone and PC are not paired, or if something else on the PC is already connected to Handsfree e.g. Microsoft Phone Link. There are other APIs on Windows 10/11 to initiate phone calls on connected devices rather than an exclusive connection to the Handsfree service.

abusaadp, 2023-11-13:
Do you have any sample code using the other APIs for the bluetooth call to work on an iPhone from Windows machine.

peterfoot, 2023-11-13:
Sorry, don't have anything as it's outside the scope of the project.

closed 2024-03-09 by peterfoot