inthehand/32feet #372: Trying to pair devices that present a pin than needs to be entered on the device.

DamoD2001, 2023-12-03:
I have successfully implemented the ability to pair devices that do not require a pin, but in testing I then tried a BT keyboard and a BT barcode scanner. In Windows there is a 6 digit pin that is shown and then I enter that and it pairs, using InTheHand.NET.Bluetooth I can't seem to use the BluetoothWin32Authentication to display and receive the entered pairing pin - is this not possible with this build? My online searching seems to indicate that it was in the 32feet.NET 3.5, but since I am using .NET 6 for this project I used this one.

thanks.

peterfoot, 2023-12-06:
Sadly this functionality is not currently exposed in the current version. Now that these pairing mechanisms are more popular the pairing code needs a rethink. I'll tag this as an enhancement but I think it's a fairly important one.

DamoD2001, 2023-12-06:
Thanks Peter, I will eagerly watch out for that update. 

Damo