inthehand/32feet #373: Phone call audio is not transferred to my computer machine when connecting an android device with Handsfree Profile.

PongManao, 2023-12-06:
I wrote a simple windows form application to connect bluetooth with an andriod phone. Everything looks good except for the call audio is not transferred from the phone to the computer as expected.
The phone call can be made by AT Commands but the audio cannot be transferred to the computer via bluetooth.
I'm not sure this is a limitation or anything is wrong with my codes.

PongManao, 2023-12-07:
> Audio is passed over a separate SCO channel which is handled by the OS at a lower level. If you're using Windows 10 there are Separate APIs for supporting hands free calling which support audio.

Thanks @peterfoot. I'm a newby in bluetooth integration. Could you please point me to those APIs?

peterfoot, 2023-12-18:
Here is the main page for the docs - https://learn.microsoft.com/en-us/uwp/api/windows.applicationmodel.calls

closed 2023-12-18 by peterfoot

n8zach, 2024-01-19:
The windows.applicationmodel.calls “Controls multiple aspects of how an application on the phone behaves”…. I don’t see how that helps or answers this question.  What am I missing?

peterfoot, 2024-01-19:
Hi, I think the namespace description is misleading and a hangover from the days of Windows Phone. The PhoneCallManager class is the entry point to the functionality along with its RequestStoreAsync method getting a PhoneCallStore which allows you to setup a watcher to enumerate phone lines, or GetDefaultLineAsync to get the default id. Use PhoneLine.FromIdAsync to create an instance of PhoneLine which will allow you to make calls.

n8zach, 2024-01-19:
Thank you for your quick and informative response.  Unfortunately I find UWP is not something I can easily support in my current software stack (see: https://stackoverflow.com/questions/75294870/use-uwp-library-in-net-framework-net-standard-application).  

Can you suggest any other native or .NET Framework approaches to relaying the audio (mic+speaker) from the phone to the PC (like an automotive head unit does)?