inthehand/32feet #376: Xamarin Mac - PlatformNotSupported exception

Archonic944, 2023-12-12:
When trying to create a new BluetoothClient, I get this exception:
`System.PlatformNotSupportedException: Operation is not supported on this platform.
  at InTheHand.Net.Sockets.BluetoothClient..ctor () [0x0000e] in <94af572180da485e94ec3b79bdd26b9c>:0
  at BrainBluetoothDetection.AppDelegate.DidFinishLaunching (Foundation.NSNotification notification) [0x00001] in /Users/gabriel/Projects/BrainBluetoothDetection/BrainBluetoothDetection/AppDelegate.cs:16
  at at (wrapper managed-to-native) AppKit.NSApplication.NSApplicationMain(int,intptr)
  at AppKit.NSApplication.Main (System.String[] args) [0x00048] in /Library/Frameworks/Xamarin.Mac.framework/Versions/9.3.0.23/src/Xamarin.Mac/AppKit/NSApplication.cs:119
  at BrainBluetoothDetection.MainClass.Main (System.String[] args) [0x00007] in /Users/gabriel/Projects/BrainBluetoothDetection/BrainBluetoothDetection/Main.cs:10
`
I am on MacOS Ventura 13.0.1. I have the NuGet packages IOBluetooth, .Net.Bluetooth, .IOBluetoothUI, and Bluetooth.Permissions installed. I'm using Visual Studio for Mac. You can have a look at my project here:
[Archive.zip](https://github.com/inthehand/32feet/files/13642876/Archive.zip)

peterfoot, 2023-12-12:
Bluetooth Classic is not currently implemented for macOS. The underlying IOBluetooth binding library is available so you can use the low level API at a push.

Archonic944, 2023-12-12:
Ah, that's unfortunate. I made a PRQ to clarify that in the README, assuming that it's available on Linux too.

Archonic944, 2023-12-12:
Reopening just to ask: do you know of any good Bluetooth libraries (C# or other lang) for making a bluetooth application from Mac? I found jSerialComm but I'm not sure if that's ideal.