inthehand/32feet #383: .NET.Bluetooth/.NET.sockets BluetoothClient throws error on Android 13

VervaekeSofie, 2023-12-26:
Hi,
I used InTheHand.Net.Bluetooth in a MAUI app and everything works on a old tablet with Android 8.1.
I tried it now on a new tablet with Android 13 and the app is interrupt on this code: BluetoothClient client = new BluetoothClient();

All permissions for bluetooth are on and also ACCESS_FINE_LOCATION.

What do i have to change, so this works also on android 13?

peterfoot, 2023-12-26:
Are you requesting permission at runtime e.g. https://inthehand.com/2023/10/11/asking-for-permission/

VervaekeSofie, 2023-12-26:
I did this know and this error is shown: Java.Lang.SecurityException: 'Need android.permission.BLUETOOTH_SCAN permission for AttributionSource { uid = 10234, packageName = com.companyname.rgbleds, attributionTag = null, token = android.os.BinderProxy@8e4298, next = null }: AdapterService getScanMode'

VervaekeSofie, 2023-12-26:
Thank you, this was the solution!

VervaekeSofie, 2023-12-29:
I redeployed my app to my tablet and now I can't access it, even with the permissions. I have to manually grant access to "nearby devices".
And all Permissions are enabled, what am I overlooking?

Salim-Farran, 2024-02-02:
Hi, did you find a solution for this issue ? I'm having the same issue with my app in release but on android 11, I'm getting the following exception on the code ( BluetoothClient client = new BluetoothClient() ): System.TypeInitializationException: The type initializer for 'InTheHand.Net.Bluetooth.AndroidBluetoothRadio' threw an exception. In debug everything works fine.
All permissions are enabled also.

Thank you.

peterfoot, 2024-03-09:
What are your linker settings? Can you exclude InTheHand.Net.Bluetooth from linking?