inthehand/32feet #385: Bluetooth LE Picker does not work on Android 11 and lower versions.

piselectrocu, 2023-12-28:
4.0.35 = not working
4.0.34 = working

I tested it with 3 different phones. There seems to be a problem with version 4.0.35. 

I'm using .net maui and .net8.0. The device selection screen opens and closes immediately.

**AndroidManifest.xml** = 
<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android">
	<uses-sdk android:minSdkVersion="21" android:targetSdkVersion="34" />
	<application android:allowBackup="true" android:icon="@mipmap/appicon" android:supportsRtl="true"></application>
	<uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
	<uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
	<uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />
	<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
</manifest>

peterfoot, 2024-01-28:
Found the error - it is incorrectly checking the status of BLUETOOTH_CONNECT permission which is not required prior to Android 12. Fix is in the code, NuGet release to follow soon.

peterfoot, 2024-03-10:
Fixed in 4.0.36

closed 2024-03-10 by peterfoot