inthehand/32feet #389: filter doesn't work using ScanForDevicesAsync on Windows10

pinibz, 2024-01-14:
Hi

I am using the code bellow on windows 10 to scan only for devices that match a known service of a specific device type.
However, it returns **all** the BLE devices around regardless of their service.

 var filter = new BluetoothLEScanFilter();
 filter.Services.Add(my_device_main_service_uuid); 
 var options = new RequestDeviceOptions();
 options.Filters.Add(filter);
  
 foreach (BluetoothDevice sd in await InTheHand.Bluetooth.Bluetooth.ScanForDevicesAsync(options))
 {
     Debug.WriteLine($"Found Device {sd.Id} {sd.Name} count {scannedDevicesList.Count()}");
 }

Am I missing something?

Thanks

matej-mnoucek, 2024-03-08:
Same issue for me. It seems that `BluetoothLEScanFilter` is simply not taken into account.

peterfoot, 2024-03-09:
Sorry, I thought this was already tracked by an issue. Currently on Windows only the NamePrefix filter is implemented for ScanForDevicesAsync, for the RequestDeviceAsync lookup it is able to build more complex filters and use service uuid. We will have to implement this in the library rather than the discovery filter.

peterfoot, 2024-03-10:
This is resolved in 4.0.36

closed 2024-03-10 by peterfoot