inthehand/32feet #390: MAUI macOS Terminating app due to uncaught exception 'System.ArgumentException', reason: 'idx (System.ArgumentException)

npdwc, 2024-01-23:
when I scan Bluetooth on macOS I get an exception
Looks like https://github.com/inthehand/32feet/issues/262 but for macOS

```
*** Terminating app due to uncaught exception 'System.ArgumentException', reason: 'idx (System.ArgumentException)
   at Foundation.NSData.get_Item(IntPtr idx) in /Users/builder/azdo/_work/1/s/xamarin-macios/src/Foundation/NSData.cs:line 280
   at InTheHand.Bluetooth.BluetoothUuid.op_Implicit(CBUUID uuid)
   at InTheHand.Bluetooth.BluetoothAdvertisingEvent..ctor(BluetoothDevice device, NSDictionary advertisementData, NSNumber RSSI)
   at InTheHand.Bluetooth.Bluetooth._manager_DiscoveredPeripheral(Object sender, CBDiscoveredPeripheralEventArgs e)
   at CoreBluetooth.CBCentralManager._CBCentralManagerDelegate.DiscoveredPeripheral(CBCentralManager central, CBPeripheral peripheral, NSDictionary advertisementData, NSNumber RSSI) in /Users/builder/azdo/_work/1/s/xamarin-macios/src/build/dotnet/maccatalyst/generated-sources/CoreBluetooth/CBCentralManager.g.cs:line 557
'
```

peterfoot, 2024-01-28:
This is possibly because the CBUUID in question is 32 bit/4 byte and the conversion logic only currently supports 2 byte short ids or the full 16 byte uuids. I've added better error handling to return Guid.Empty if the length is not recognised, but will need to add 32 bit conversion too.

mksdevelop, 2024-02-29:
I'm getting this error on ios.
Maybe we should wrap some of the code in a try catch block? Or release a library update?

```
Unhandled Exception:
System.ArgumentException: idx
   at Foundation.NSData.get_Item(IntPtr idx) in /Users/builder/azdo/_work/1/s/xamarin-macios/src/Foundation/NSData.cs:line 280
   at InTheHand.Bluetooth.BluetoothUuid.op_Implicit(CBUUID uuid)
   at InTheHand.Bluetooth.BluetoothAdvertisingEvent..ctor(BluetoothDevice device, NSDictionary advertisementData, NSNumber RSSI)
   at InTheHand.Bluetooth.Bluetooth._manager_DiscoveredPeripheral(Object sender, CBDiscoveredPeripheralEventArgs e)
   at CoreBluetooth.CBCentralManager._CBCentralManagerDelegate.DiscoveredPeripheral(CBCentralManager central, CBPeripheral peripheral, NSDictionary advertisementData, NSNumber RSSI) in /Users/builder/azdo/_work/1/s/xamarin-macios/src/build/dotnet/ios/generated-sources/CoreBluetooth/CBCentralManager.g.cs:line 569
   at UIKit.UIApplication.UIApplicationMain(Int32 argc, String[] argv, IntPtr principalClassName, IntPtr delegateClassName) in /Users/builder/azdo/_work/1/s/xamarin-macios/src/UIKit/UIApplication.cs:line 58
   at UIKit.UIApplication.Main(String[] args, Type principalClass, Type delegateClass) in /Users/builder/azdo/_work/1/s/xamarin-macios/src/UIKit/UIApplication.cs:line 94
   at Project1.Program.Main(String[] args) in /Users/dev/Develop/Projects/Project1 Project1/Platforms/iOS/Program.cs:line 13
2024-02-29 11:46:20.258 Project1[1189:614552] Unhandled managed exception: idx (System.ArgumentException)
   at Foundation.NSData.get_Item(IntPtr idx) in /Users/builder/azdo/_work/1/s/xamarin-macios/src/Foundation/NSData.cs:line 280
   at InTheHand.Bluetooth.BluetoothUuid.op_Implicit(CBUUID uuid)
   at InTheHand.Bluetooth.BluetoothAdvertisingEvent..ctor(BluetoothDevice device, NSDictionary advertisementData, NSNumber RSSI)
   at InTheHand.Bluetooth.Bluetooth._manager_DiscoveredPeripheral(Object sender, CBDiscoveredPeripheralEventArgs e)
   at CoreBluetooth.CBCentralManager._CBCentralManagerDelegate.DiscoveredPeripheral(CBCentralManager central, CBPeripheral peripheral, NSDictionary advertisementData, NSNumber RSSI) in /Users/builder/azdo/_work/1/s/xamarin-macios/src/build/dotnet/ios/generated-sources/CoreBluetooth/CBCentralManager.g.cs:line 569
   at UIKit.UIApplication.UIApplicationMain(Int32 argc, String[] argv, IntPtr principalClassName, IntPtr delegateClassName) in /Users/builder/azdo/_work/1/s/xamarin-macios/src/UIKit/UIApplication.cs:line 58
   at UIKit.UIApplication.Main(String[] args, Type principalClass, Type delegateClass) in /Users/builder/azdo/_work/1/s/xamarin-macios/src/UIKit/UIApplication.cs:line 94
   at Project1.Program.Main(String[] args) in /Users/dev/Develop/Projects/Project1/Project1/Platforms/iOS/Program.cs:line 13
```