inthehand/32feet #394: Bluetooth.RequestDeviceAsync() Not show UI "List with awaliable devices"

anonimysUA, 2024-02-07:
Im write Windows Console Application, try to call code below:

```
BluetoothDevice device = null;
Task.Run(async () =>
{
    device = await Bluetooth.RequestDeviceAsync();

}).GetAwaiter().GetResult();
```

Result is null without show UI "List with awaliable devices". For fix that im always need a few time make a Clear solution and make a new rebuild for work.
What i need a do for show UI each time when call Bluetooth.RequestDeviceAsync()?

Im use :
- Win 11 Pro
- .Net Framework 4.6.1. 
- InTheHand.Net.Bluetooth 4.1.0.0 (Runtime v4.0.30319)

peterfoot, 2024-02-07:
Is there any output in the debug window to indicate an error when you call RequestDeviceAsync? My guess is it is the code to get the HWND for the console window as this is more problematic than a desktop application.

anonimysUA, 2024-02-07:
I see in VS output that information:

'BluetoothWin.exe' (CLR v4.0.30319: DefaultDomain): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_64\mscorlib\v4.0_4.0.0.0__b77a5c561934e089\mscorlib.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'BluetoothWin.exe' (CLR v4.0.30319: DefaultDomain): Loaded 'D:...\BLE_Test\BluetoothWin\bin\Debug\BluetoothWin.exe'. Symbols loaded.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'D:...\BLE_Test\BluetoothWin\bin\Debug\InTheHand.BluetoothLE.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Step into: Stepping over non-user code 'BluetoothWin.Program.Main'
Step into: Stepping over non-user code 'BluetoothWin.Program.<Main>d__2..ctor'
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'D:...\BLE_Test\BluetoothWin\bin\Debug\InTheHand.Net.Bluetooth.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Core\v4.0_4.0.0.0__b77a5c561934e089\System.Core.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System\v4.0_4.0.0.0__b77a5c561934e089\System.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.

**// Here start Run Task Bluetooth.RequestDeviceAsync()**

'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\mscorlib.resources\v4.0_4.0.0.0_ru_b77a5c561934e089\mscorlib.resources.dll'. Module was built without symbols.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\system32\WinMetadata\Windows.Devices.winmd'. Module was built without symbols.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Runtime\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.Runtime.dll'. Module was built without symbols.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\system32\WinMetadata\Windows.Foundation.winmd'. Module was built without symbols.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Runtime.InteropServices.WindowsRuntime\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.Runtime.InteropServices.WindowsRuntime.dll'. Module was built without symbols.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Runtime.WindowsRuntime\v4.0_4.0.0.0__b77a5c561934e089\System.Runtime.WindowsRuntime.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\system32\WinMetadata\Windows.UI.winmd'. Module was built without symbols.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\system32\WinMetadata\Windows.ApplicationModel.winmd'. Module was built without symbols.

**// Result is back**

The thread 0x666c has exited with code 0 (0x0).
The thread 0xb34c has exited with code 0 (0x0).

anonimysUA, 2024-02-07:
Output above was represent a fail behavior.

**For comparison below is Sucess call** RequestDeviceAsync(), with open UI to choose avaliable devices, and with choose 1 by click from list

'BluetoothWin.exe' (CLR v4.0.30319: DefaultDomain): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_64\mscorlib\v4.0_4.0.0.0__b77a5c561934e089\mscorlib.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'BluetoothWin.exe' (CLR v4.0.30319: DefaultDomain): Loaded 'D:...\BLE_Test\BluetoothWin\bin\Debug\BluetoothWin.exe'. Symbols loaded.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'D:...\BLE_Test\BluetoothWin\bin\Debug\InTheHand.BluetoothLE.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Step into: Stepping over non-user code 'BluetoothWin.Program.Main'
Step into: Stepping over non-user code 'BluetoothWin.Program.<Main>d__2..ctor'
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'D:...\BLE_Test\BluetoothWin\bin\Debug\InTheHand.Net.Bluetooth.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Core\v4.0_4.0.0.0__b77a5c561934e089\System.Core.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System\v4.0_4.0.0.0__b77a5c561934e089\System.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\mscorlib.resources\v4.0_4.0.0.0_ru_b77a5c561934e089\mscorlib.resources.dll'. Module was built without symbols.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\system32\WinMetadata\Windows.Devices.winmd'. Module was built without symbols.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Runtime\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.Runtime.dll'. Module was built without symbols.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\system32\WinMetadata\Windows.Foundation.winmd'. Module was built without symbols.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Runtime.InteropServices.WindowsRuntime\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.Runtime.InteropServices.WindowsRuntime.dll'. Module was built without symbols.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Runtime.WindowsRuntime\v4.0_4.0.0.0__b77a5c561934e089\System.Runtime.WindowsRuntime.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\system32\WinMetadata\Windows.UI.winmd'. Module was built without symbols.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\system32\WinMetadata\Windows.ApplicationModel.winmd'. Module was built without symbols.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\Remote Debugger\x64\Runtime\Microsoft.VisualStudio.Debugger.Runtime.Desktop.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Configuration\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.Configuration.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'BluetoothWin.exe' (CLR v4.0.30319: BluetoothWin.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Xml\v4.0_4.0.0.0__b77a5c561934e089\System.Xml.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.

igiona, 2024-06-18:
I'm not sure if it's related, but in my UI-based application the `RequestDeviceAsync` needs to be executed from within the UI-thread in order to work.

Now, I'm not sure what is the "ui-thread" of a console application, probably simply the main thread.
Hence packing the call in a Task it's surely not going to run in the main thread.
Maybe this is causing you troubles?

MulleDK19, 2025-02-10:
Same problem, Android 14. No device list comes up, it just returns null.

The only way I can get it to pop up and show is to first attempt `await Bluetooth.RequestDeviceAsync();`, then open the bluetooth UI in the OS, and click "Pair new device", the moment the system's list pops up, I can switch back to my app, try to connect again, and then the menu pops up with devices and I can connect.

```csharp
BluetoothDevice device = null;
bool isAvailable = await Bluetooth.GetAvailabilityAsync();
if (isAvailable)
{
    RequestDeviceOptions rdo = new RequestDeviceOptions();
    rdo.AcceptAllDevices = true;
    device = await Bluetooth.RequestDeviceAsync(rdo);
}
```