inthehand/32feet #395: How to to connect 2 Bluetooth Serial Ports on the same device

matthiasbaudot, 2024-02-07:
Hello, 
I'm developing an application that needs to connect to 2 virtual COM ports on a Bluetooth Device.
Those 2 serial ports are exposed as Services with the same GUID (00001101-0000-1000-8000-00805f9b34fb) on 2 different channels.
When I use the SetServiceState method to create the virtual COM port, I can only pass the service GUID (not the channel), which creates one of the 2 COM ports. 
How can I create the 2nd COM port?
Thanks for your help.

matthiasbaudot, 2024-02-08:
@peterfoot I see you wrote several wiki articles on that topic. Do you maybe have an idea on how to proceed with this use case?
Thank you!

matthiasbaudot, 2024-02-08:
Thank you @peterfoot. My application uses a 3rd party library to control the device through Serial Commands, addressing a virtual COM port. Will that work with the "Connect" method?
I just gave it a try but it doesn't seem to create any Virtual COM port after calling the "Connect" method.

peterfoot, 2024-02-08:
No sorry, that only works for direct socket access. I can't think of a workaround for virtual COM ports.

closed 2024-02-09 by matthiasbaudot