inthehand/32feet #396: object GattCharacteristic.Value crash with 'System.ArgumentException'

anonimysUA, 2024-02-07:
Im make Windows Console Application and try communicate with BLE device by UART (by TX-trancieve RX-recieve chanels), but have 2 Issue for get TX result:
1) Event CharacteristicValueChanged never triggered
2) tx_char.Value crash (same with call ReadValueAsync())

```
IReadOnlyList<GattCharacteristic> characteristics = null;
GattCharacteristic rx_char = null;
GattCharacteristic tx_char = null;

// I. Get a Service
Task.Run(async () =>
{
    List<GattService> availableServices = await token.device.Gatt.GetPrimaryServicesAsync(null);// .GetPrimaryServiceAsync(GattServiceUuids.Battery);
    rx_service = availableServices.First(x => x.Uuid.Value == RX_SERVICE_UUID);
}).GetAwaiter().GetResult();

// II. Get a characteristic
Task.Run(async () =>
{
    characteristics = await rx_service.GetCharacteristicsAsync();                         
}).GetAwaiter().GetResult();

for (int i = 0; i < characteristics.Count; i++)
{
    if(characteristics[i].Uuid.Value.Equals(ST338BLE_Const.RX_CHAR_UUID))
    {
        rx_char = characteristics[i];
        rx_char.CharacteristicValueChanged += RX_CharacteristicValueChanged;
        //rx_char.StartNotificationsAsync();
    }
    else if(characteristics[i].Uuid.Value.Equals(ST338BLE_Const.TX_CHAR_UUID))
    {
        tx_char = characteristics[i];
        tx_char.CharacteristicValueChanged += TX_CharacteristicValueChanged;
        tx_char.StartNotificationsAsync();
    }
}   

// III. Send
Task.Run(async () => 
{
    await rx_char.WriteValueWithResponseAsync(new byte[]{ /*APDU command*/ });    
}).GetAwaiter().GetResult();

// IV. Try to Receive data
                    
byte[] rec = tx_char.Value; // Crash when try read a result    

// The same Crash result
//Task.Run(async () =>
//{
//    rec = await tx_char.ReadValueAsync();
//}).GetAwaiter().GetResult();
```

Additional info:
Exception thrown 'System.ArgumentException'

StackTrace
"at System.Runtime.InteropServices.WindowsRuntime.WindowsRuntimeBufferExtensions.ToArray(IBuffer source, UInt32 sourceIndex, Int32 count)\r\n
   at InTheHand.Bluetooth.GattCharacteristic.PlatformGetValue()\r\n
   ..."

- Win 11 Pro
- .Net Framework 4.6.1.
- InTheHand.Net.Bluetooth 4.1.0.0 (Runtime v4.0.30319)

peterfoot, 2024-02-07:
can you query the Properties of both characteristics to confirm what features they support?

anonimysUA, 2024-02-08:
> can you query the Properties of both characteristics to confirm what features they support?

Im use that UUID of service/tx/rx on my native iOS/Android libs, so im sure they are supported and work.

Maybe my code belowe wrong for a get answer from device? 

```
Task.Run(async () =>
{
    rec = await tx_char.ReadValueAsync();
}).GetAwaiter().GetResult();
```

How a right way i can get result after send a APDU command to device?

cogree, 2024-07-23:
I was able to get reproduce this error; it seems related to characteristics that produce a zero byte response.

commit 80a78171 (2025-03-06): Short circuit calls to '.ToArray() on empty buffers to avoid exceptions. Fixes #396. (#453)
closed 2025-03-06 by peterfoot