inthehand/32feet #397: 32feet.NET BluetoothListener In C#

TheMajidHeydari, 2024-02-09:
I have a windows application with version .net framework 4.7.2 and I need a Bluetooth listener for receive and save files were sending to this device by Bluetooth.
I using 32feet.NET and I was written this code :
but its have some problems. if another way recommended, please tell me. other wise please help me to solve current problems.
it was took so long when I reading stream bytes to write into file. and after written, It was wrong data written into file :(
how I can get extension of receiving file?

```
public Form1()
{
    InitializeComponent();

    _bluetoothListener = new BluetoothListener(BluetoothService.ObexObjectPush);
    _listenerThread = new Thread(ListenForBluetoothRequests);
    _listenerThread.Start();
}

private void ListenForBluetoothRequests()
{
    _bluetoothListener.Start();

    while (true)
    {
        try
        {
            _bluetoothListener.Pending();
            BluetoothClient client = _bluetoothListener.AcceptBluetoothClient();

            if (client != null)
            {
                // Handle the incoming file transfer
                HandleIncomingFile(client);

                // Close the client connection
                client.Close();
            }
        }
        catch (IOException ex)
        {
            // Handle exceptions, if any
            Console.WriteLine("Exception: " + ex.Message);
        }
    }
}

private void HandleIncomingFile(BluetoothClient client)
{
    // Customize the file save path as per your requirement
    string savePath = "C:\\BluetoothReceivedFiles\\";

    if (!Directory.Exists(savePath))
    {
        Directory.CreateDirectory(savePath);
    }

    Stream networkStream = client.GetStream();
    byte[] buffer = new byte[1024];
    int bytesRead;

    // Read the incoming file and save it
    using (FileStream fileStream = File.Create(Path.Combine(savePath, "ReceivedFile.txt")))
    {
        while ((bytesRead = networkStream.Read(buffer, 0, buffer.Length)) > 0)
        {
            fileStream.Write(buffer, 0, bytesRead);
        }
    }
}
```

peterfoot, 2024-02-09:
To receive files over OBEX, use the InTheHand.Net.Obex library and the ObexListener class. The BluetoothListener will only get the raw packets and you'll have to read the headers and send the correct responses to each packet.

TheMajidHeydari, 2024-02-09:
> To receive files over OBEX, use the InTheHand.Net.Obex library and the ObexListener class. The BluetoothListener will only get the raw packets and you'll have to read the headers and send the correct responses to each packet.

thanks. 
i want to receive files over bluetooth. is obex handler work for that ?

closed 2024-02-09 by TheMajidHeydari