inthehand/32feet #398: Win11 C# - BluetoothRadio.Default 'Operation is not supported on this platform.'

SomeYahoo, 2024-02-21:
Microsoft Visual Studio Community 2022
Version 17.8.3
VisualStudio.17.Release/17.8.3+34330.188
Microsoft .NET Framework
Version 4.8.09032

Installed Version: Community

Windows 11 Pro

InTheHand.Net.Bluetooth version 4.1.40 from NuGet (latest version)

**BluetoothRadio.Default** yields **'Operation is not supported on this platform.'**
...and returns null.
 
ChatGPT seems to think I need **BluetoothRadio.PrimaryRadio**, but this is not recognized.

SomeYahoo, 2024-02-21:
What I'm trying to do is find the default radio, and enable it if it's off.

peterfoot, 2024-02-22:
This tends to occur when your project doesn't target a specific enough TFM and the .NET Standard dll is used from the NuGet package. The consuming package needs to target net6.0-windows7.0 or net6.0-windows10.0.19041.0 or higher to pick up the Win32 or WinRT implementations respectively.

PerKjelsvik, 2024-02-22:
version: [`InTheHand.Net.Bluetooth 4.1.40`](https://www.nuget.org/packages/InTheHand.Net.Bluetooth/4.1.40)
Framework: WPF

Just to chime in here: 

I reference [`WinRTNetworkStream`](https://github.com/inthehand/32feet/blob/f54f691d9c3b11ac742b3f14ef4e6d4ce95b1d0d/InTheHand.Net.Bluetooth/Platforms/Windows/WinRTNetworkStream.Windows.cs) directly in one of mye Bluetooth classes. If I target `net6.0-windows10.0.19041.0` I can't compile the application:

> [`CS0246`](https://learn.microsoft.com/en-us/dotnet/csharp/language-reference/compiler-messages/cs0246): The type or namespace name 'WinRTNetworkStream' could not be found (are you missing a using directive or an assembly reference?)

So it seems that targeting `net6.0-windows10.0.19041.0` forces it to Win32, not WinRT. The same happens when I target `net6.0-windows10.0.20348.0`. I am able to reference it when targeting `net6.0-windows10.0.22000.0` however.

Based on that it would seem that Win32 is used for all versions before 10.0.22000.0?

If it's supposed to be for 10.0.19041.0 and up then maybe something is wrong with the package? I can reproduce it if I create a new blank WPF project. For my use case it would be great if 10.0.19041.0 and up used WinRT. If you'd like I can report it as a new issue if it's supposed to use WinRT.

peterfoot, 2024-02-22:
@PerKjelsvik any reason you need WinRTNetworkStream over NetworkStream?

PerKjelsvik, 2024-02-22:
> @PerKjelsvik any reason you need WinRTNetworkStream over NetworkStream?

No, not really! I have only used it in my class to be able to easily navigate directly to the platform-specific NetworkStream. Normally I would just use `NetworkStream`, but when I explored using the WinRT version in November (#361) I referenced it directly to look into the implementation, and that's also how I discovered it didn't work for build version 10.0.19041.0.

> For .NET 6 you should be getting WinRT as long as your target SDK is 19041 or later.

Hmm, that doesn't seem to be the case? Assuming WinRTNetworkStream should be accessible when targeting `net6.0-10.0.19041.0` in the application project file. Here's what I have in a new blank WPF project:

```xml
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <OutputType>WinExe</OutputType>
    <TargetFramework>net6.0-windows10.0.19041.0</TargetFramework>
    <RootNamespace>_32FeetTest</RootNamespace>
    <Nullable>enable</Nullable>
    <ImplicitUsings>enable</ImplicitUsings>
    <UseWPF>true</UseWPF>
    <SupportedOSPlatformVersion>10.0.19041.0</SupportedOSPlatformVersion>
  </PropertyGroup>

  <ItemGroup>
    <PackageReference Include="InTheHand.Net.Bluetooth" Version="4.1.40" />
  </ItemGroup>

</Project>
```

In code:

```csharp
using InTheHand.Net.Sockets;

BluetoothClient client = new();
var stream = client.GetStream();
var winRT = (WinRTNetworkStream)stream; // can't compile, CS0246
```

Inspecting the `BluetoothClient` I also see this:

[image]

In this case the compiler flag `WINDOWS10_0_17763_0_OR_GREATER` should have triggered, right?

Or is there something wrong with my setup?