inthehand/32feet #400: You can no longer set Client.InquiryLength on Android

snydesc, 2024-02-28:
On Android we have been setting the client's Inquiry Length to 30 seconds.  It appears with the a recent update the 
Inquiry length no longer works on Android.  Looking at the AndroidBluetoothClient code it appears the Get for InquiryLength always returns 0 and the Set for  InquiryLength throws Operation is not supported on this platform

How can we set the InquiryLength on Android ?

```
Operation is not supported on this platform.
[DOTNET] 
[DOTNET]    at InTheHand.Net.Sockets.AndroidBluetoothClient.InTheHand.Net.Sockets.IBluetoothClient.set_InquiryLength(TimeSpan value)
[DOTNET]    at InTheHand.Net.Sockets.BluetoothClient.set_InquiryLength(TimeSpan value)
```

peterfoot, 2024-03-03:
While it didn't used to throw an exception, InquiryLength was never supported on Android and was a no-op. You can use the newer DiscoverDevicesAsync method on .NET 6.0 and above which allows you to pass a cancellation token so that you can stop discovery once you have the required device or after a set time.

SittenSpynne, 2025-04-23:
Maybe [the documentation](https://inthehand.github.io/html/P_InTheHand_Net_Sockets_BluetoothClient_InquiryLength.htm) needs an update, or the property could be marked with `ObsoleteAttribute`?