inthehand/32feet #402: The Bluetooth device name is not displayed on Android 14

mksdevelop, 2024-03-12:
If you connect to a Bluetooth device on Android 14 and later disconnect, the device will no longer display its name during scanning.

When I added the Shiny.BluetoothLE library to the project, it also did not work, but if I change the scanning parameter to LowLatency, it starts scanning device names.
If I run both scans at the same time, the scan in InTheHand.Bluetooth works as expected.

It seems that LowLatency somehow affects the performance of Ble on Android 14.

Can you add this setting to your library or fix scanning on Android 14?

vitalii-vov, 2024-05-18:
Все еще есть эта проблема.
После отключения от устройства с помощью метода `_device.Gatt.Disconnect();` имя устройства больше не обнаруживается. 
Android 14
Pixel 6a

```
private void OnAdvertisementReceived(object sender, BluetoothAdvertisingEvent e)
{
    if (e.Device is null)
        return;
    Debug.WriteLine($"{e.Id} {e.Name}");
}
```
log
```
90:FD:9F:14:39:B9 tes4me-0000    <-- NAME IS EXIST
90:FD:9F:14:39:B9 tes4me-0000    <-- NAME IS EXIST
E0:03:6B:D9:80:45 Samsung S95CA 65
70:28:45:5F:75:3C Oclean Bluetooth
90:FD:9F:14:39:B9 tes4me-0000    <-- NAME IS EXIST
90:FD:9F:14:39:B9 tes4me-0000    <-- NAME IS EXIST
[BluetoothGatt] connect() - device: 90:FD:9F:14:39:B9, auto: false
[BluetoothGatt] registerApp()
[BluetoothGatt] registerApp() - UUID=b4a1c494-663f-43b1-b737-f72a9f751fcf
[BluetoothGatt] connect(void) - device: 90:FD:9F:14:39:B9, auto=false
[BluetoothGatt] onClientRegistered() - status=0 clientIf=6

[BluetoothGatt] onClientConnectionState() - status=0 clientIf=6 device=90:FD:9F:14:39:B9
[BluetoothGatt] discoverServices() - device: 90:FD:9F:14:39:B9
24:FC:E5:86:E3:57 [TV] Samsung The Frame (55)
2C:4C:C6:4A:25:23 Venus_2C4CC64A2523
[BluetoothGatt] onConnectionUpdated() - Device=90:FD:9F:14:39:B9 interval=12 latency=0 timeout=500 status=0
[BluetoothGatt] onConnectionUpdated() - Device=90:FD:9F:14:39:B9 interval=6 latency=0 timeout=500 status=0

[BluetoothGatt] onSearchComplete() = Device=90:FD:9F:14:39:B9 Status=0
[BluetoothGatt] onConnectionUpdated() - Device=90:FD:9F:14:39:B9 interval=12 latency=0 timeout=500 status=0
[BluetoothAdapter] isLeEnabled(): ON

[BluetoothGatt] cancelOpen() - device: 90:FD:9F:14:39:B9
[BluetoothGatt] onClientConnectionState() - status=0 clientIf=6 device=90:FD:9F:14:39:B9
[BluetoothAdapter] isLeEnabled(): ON
[BluetoothLeScanner] onScannerRegistered() - status=0 scannerId=7 mScannerId=0
E0:03:6B:D9:80:45 Samsung S95CA 65
70:28:45:5F:75:3C Oclean Bluetooth
38:5B:44:3E:D6:7A Square Reader 4510
41:5E:8D:5A:DF:9D 
90:FD:9F:14:39:B9     <-- NAME IS NOT EXIST
2C:4C:C6:4A:25:23 Venus_2C4CC64A2523
24:FC:E5:86:E3:57 [TV] Samsung The Frame (55)
90:FD:9F:14:39:B9     <-- NAME IS NOT EXIST
44:5C:E9:EB:A0:4F 
```

vitalii-vov, 2024-05-18:
Still have this problem.
After disconnecting from the device using the `_device.Gatt.Disconnect();` method, the device name is no longer detected. 
Android 14
Pixel 6a
InTheHand.BluetoothLE 4.0.36

```
private void OnAdvertisementReceived(object sender, BluetoothAdvertisingEvent e)
{
    if (e.Device is null)
        return;
    Debug.WriteLine($"{e.Id} {e.Name}");
}
```
log
```
90:FD:9F:14:39:B9 tes4me-0000    <-- NAME IS EXIST
90:FD:9F:14:39:B9 tes4me-0000    <-- NAME IS EXIST
E0:03:6B:D9:80:45 Samsung S95CA 65
70:28:45:5F:75:3C Oclean Bluetooth
90:FD:9F:14:39:B9 tes4me-0000    <-- NAME IS EXIST
90:FD:9F:14:39:B9 tes4me-0000    <-- NAME IS EXIST
[BluetoothGatt] connect() - device: 90:FD:9F:14:39:B9, auto: false
[BluetoothGatt] registerApp()
[BluetoothGatt] registerApp() - UUID=b4a1c494-663f-43b1-b737-f72a9f751fcf
[BluetoothGatt] connect(void) - device: 90:FD:9F:14:39:B9, auto=false
[BluetoothGatt] onClientRegistered() - status=0 clientIf=6

[BluetoothGatt] onClientConnectionState() - status=0 clientIf=6 device=90:FD:9F:14:39:B9
[BluetoothGatt] discoverServices() - device: 90:FD:9F:14:39:B9
24:FC:E5:86:E3:57 [TV] Samsung The Frame (55)
2C:4C:C6:4A:25:23 Venus_2C4CC64A2523
[BluetoothGatt] onConnectionUpdated() - Device=90:FD:9F:14:39:B9 interval=12 latency=0 timeout=500 status=0
[BluetoothGatt] onConnectionUpdated() - Device=90:FD:9F:14:39:B9 interval=6 latency=0 timeout=500 status=0

[BluetoothGatt] onSearchComplete() = Device=90:FD:9F:14:39:B9 Status=0
[BluetoothGatt] onConnectionUpdated() - Device=90:FD:9F:14:39:B9 interval=12 latency=0 timeout=500 status=0
[BluetoothAdapter] isLeEnabled(): ON

[BluetoothGatt] cancelOpen() - device: 90:FD:9F:14:39:B9
[BluetoothGatt] onClientConnectionState() - status=0 clientIf=6 device=90:FD:9F:14:39:B9
[BluetoothAdapter] isLeEnabled(): ON
[BluetoothLeScanner] onScannerRegistered() - status=0 scannerId=7 mScannerId=0
E0:03:6B:D9:80:45 Samsung S95CA 65
70:28:45:5F:75:3C Oclean Bluetooth
38:5B:44:3E:D6:7A Square Reader 4510
41:5E:8D:5A:DF:9D 
90:FD:9F:14:39:B9     <-- NAME IS NOT EXIST
2C:4C:C6:4A:25:23 Venus_2C4CC64A2523
24:FC:E5:86:E3:57 [TV] Samsung The Frame (55)
90:FD:9F:14:39:B9     <-- NAME IS NOT EXIST
44:5C:E9:EB:A0:4F 
```