inthehand/32feet #405: Xamarin.iOS - Client.PairedDevices does not return the devices

Ekta-M-Patel, 2024-03-19:
Hi,

I am using this library to get a list of paired devices In my Xamarin.Forms app. I can get a device list in Android but not in iOS. 
when I call to BluetoothClient.PairedDevices, it returns values when on Android, but on iOS it's always empty. 

Does this library have support for iOS or is there any mistake in the code?

I am attaching the code for reference.

`BluetoothClient client = new BluetoothClient();
foreach (var device in client.PairedDevices)
{
GeolocationDevices.Add(device.DeviceName);
}`

peterfoot, 2024-03-28:
I think that this is because the iOS API is a bit different in that it shows devices previously connected by the app, rather than paired to the device. If it's not showing a device that you have previously successfully connected to please let me know.

Ekta-M-Patel, 2024-03-29:
@peterfoot, It doesn't show the previously successfully connected device also. iOS always return a null/empty value in BluetoothClient.PairedDevices field.