inthehand/32feet #409: [MACOS BLE]Tmds.DBus.ConnectException: Can't assign requested address

SydneyOwl, 2024-04-14:
Hello,
When calling either `ScanForDevicesAsync` or `RequestDeviceAsync` on macOS 13.5 a `Tmds.DBus.ConnectException` will always occur, 
Here's the examle code:
```c#
var cts = new CancellationTokenSource();
cts.CancelAfter(5000);
var discoveredDevices = await Bluetooth.ScanForDevicesAsync(new RequestDeviceOptions()
{
    Filters = { filter }
},cts.Token);
foreach (var discoveredDevice in discoveredDevices)
{
    if (discoveredDevice.Name.Equals(BLE.BTNAME_SHX8800))
    {
        device = discoveredDevice;
        break;
    }
}
```
and the complete stack trace:
```sh
Unhandled exception. Tmds.DBus.ConnectException: Can't assign requested address
 ---> System.Net.Sockets.SocketException (49): Can't assign requested address
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.CreateException(SocketError error, Boolean forAsyncThrow)
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.ConnectAsync(Socket socket)
   at System.Net.Sockets.Socket.ConnectAsync(EndPoint remoteEP, CancellationToken cancellationToken)
   at System.Net.Sockets.Socket.ConnectAsync(EndPoint remoteEP)
   at Tmds.DBus.Transports.TransportSocket.ConnectAsync(EndPoint endPoint)
   at Tmds.DBus.Transports.TransportSocket.ConnectUnixAsync(AddressEntry entry, CancellationToken cancellationToken, Boolean supportsFdPassing)
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.Start[TStateMachine](TStateMachine& stateMachine)
   at Tmds.DBus.Transports.TransportSocket.ConnectUnixAsync(AddressEntry entry, CancellationToken cancellationToken, Boolean supportsFdPassing)
   at Tmds.DBus.Transports.TransportSocket.ConnectAsync(AddressEntry entry, CancellationToken cancellationToken, Boolean supportsFdPassing)
   at Tmds.DBus.Transports.Transport.ConnectAsync(AddressEntry entry, ClientSetupResult connectionContext, CancellationToken cancellationToken)
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.Start[TStateMachine](TStateMachine& stateMachine)
   at Tmds.DBus.Transports.Transport.ConnectAsync(AddressEntry entry, ClientSetupResult connectionContext, CancellationToken cancellationToken)
   at Tmds.DBus.DBusConnection.ConnectAsync(ClientSetupResult connectionContext, Boolean runContinuationsAsynchronously, Action`1 onDisconnect, CancellationToken cancellationToken)
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.Start[TStateMachine](TStateMachine& stateMachine)
   at Tmds.DBus.DBusConnection.ConnectAsync(ClientSetupResult connectionContext, Boolean runContinuationsAsynchronously, Action`1 onDisconnect, CancellationToken cancellationToken)
   at Tmds.DBus.Connection.DoConnectAsync()
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.Start[TStateMachine](TStateMachine& stateMachine)
   at Tmds.DBus.Connection.DoConnectAsync()
   at Tmds.DBus.Connection.GetConnectionTask()
   at Tmds.DBus.Connection.CallMethodAsync(Message message)
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.Start[TStateMachine](TStateMachine& stateMachine)
   at Tmds.DBus.Connection.CallMethodAsync(Message message)
   at Tmds.DBus.CodeGen.DBusObjectProxy.SendMethodReturnReaderAsync(String iface, String member, Nullable`1 inSignature, MessageWriter writer)
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.Start[TStateMachine](TStateMachine& stateMachine)
   at Tmds.DBus.CodeGen.DBusObjectProxy.SendMethodReturnReaderAsync(String iface, String member, Nullable`1 inSignature, MessageWriter writer)
   at Tmds.DBus.CodeGen.DBusObjectProxy.CallNonVoidMethodAsync[T](String iface, String member, Nullable`1 inSignature, MessageWriter writer, ReadMethodDelegate`1 readValue)
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.Start[TStateMachine](TStateMachine& stateMachine)
   at Tmds.DBus.CodeGen.DBusObjectProxy.CallNonVoidMethodAsync[T](String iface, String member, Nullable`1 inSignature, MessageWriter writer, ReadMethodDelegate`1 readValue)
   at Linux.Bluetooth.IObjectManagerProxy.GetManagedObjectsAsync()
   at Linux.Bluetooth.BlueZManager.GetProxiesAsync[T](String interfaceName, IDBusObject rootObject)
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.Start[TStateMachine](TStateMachine& stateMachine)
   at Linux.Bluetooth.BlueZManager.GetProxiesAsync[T](String interfaceName, IDBusObject rootObject)
   at Linux.Bluetooth.BlueZManager.GetAdaptersAsync()
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.Start[TStateMachine](TStateMachine& stateMachine)
   at Linux.Bluetooth.BlueZManager.GetAdaptersAsync()
   at InTheHand.Bluetooth.Bluetooth.Initialize()
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.Start[TStateMachine](TStateMachine& stateMachine)
   at InTheHand.Bluetooth.Bluetooth.Initialize()
   at InTheHand.Bluetooth.Bluetooth.PlatformScanForDevices(RequestDeviceOptions options, CancellationToken cancellationToken)
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.Start[TStateMachine](TStateMachine& stateMachine)
   at InTheHand.Bluetooth.Bluetooth.PlatformScanForDevices(RequestDeviceOptions options, CancellationToken cancellationToken)
   at InTheHand.Bluetooth.Bluetooth.ScanForDevicesAsync(RequestDeviceOptions options, CancellationToken cancellationToken)
   at shx8x00.Views.MainWindow.MenuConnectBT_OnClick(Object sender, RoutedEventArgs e) in /Users/ruisu/Desktop/shx8x00_universal/Views/MainWindow.axaml.cs:line 481
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.Start[TStateMachine](TStateMachine& stateMachine)
   at shx8x00.Views.MainWindow.MenuConnectBT_OnClick(Object sender, RoutedEventArgs e)
   at Avalonia.Interactivity.EventRoute.RaiseEventImpl(RoutedEventArgs e)
   at Avalonia.Interactivity.EventRoute.RaiseEvent(Interactive source, RoutedEventArgs e)
[... 31 lines cut ...]
   at Tmds.DBus.CodeGen.DBusObjectProxy.CallNonVoidMethodAsync[T](String iface, String member, Nullable`1 inSignature, MessageWriter writer, ReadMethodDelegate`1 readValue)
   at Linux.Bluetooth.BlueZManager.GetProxiesAsync[T](String interfaceName, IDBusObject rootObject)
   at Linux.Bluetooth.BlueZManager.GetAdaptersAsync()
   at InTheHand.Bluetooth.Bluetooth.Initialize()
   at InTheHand.Bluetooth.Bluetooth.PlatformScanForDevices(RequestDeviceOptions options, CancellationToken cancellationToken)
   at shx8x00.Views.MainWindow.MenuConnectBT_OnClick(Object sender, RoutedEventArgs e) in /Users/ruisu/Desktop/shx8x00_universal/Views/MainWindow.axaml.cs:line 481
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.SendOrPostCallbackDispatcherOperation.InvokeCore()
   at Avalonia.Threading.DispatcherOperation.Execute()
   at Avalonia.Threading.Dispatcher.ExecuteJob(DispatcherOperation job)
   at Avalonia.Threading.Dispatcher.ExecuteJobsCore(Boolean fromExplicitBackgroundProcessingCallback)
   at Avalonia.Threading.Dispatcher.Signaled()
   at Avalonia.Native.Interop.Impl.__MicroComIAvnPlatformThreadingInterfaceEventsVTable.Signaled(Void* this)
--- End of stack trace from previous location ---
   at Avalonia.Native.DispatcherImpl.RunLoop(CancellationToken token)
   at Avalonia.Threading.DispatcherFrame.Run(IControlledDispatcherImpl impl)
   at Avalonia.Threading.Dispatcher.PushFrame(DispatcherFrame frame)
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken)
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args)
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime(AppBuilder builder, String[] args, Action`1 lifetimeBuilder)
   at shx8x00.Program.Main(String[] args) in /Users/ruisu/Desktop/shx8x00_universal/Program.cs:line 16
```
and using sudo to run the program yields the same result....however this code runs without any issues on both Windows and Linux.

After thoroughly examining relevant documentation and examples, I have been unable to pinpoint the error in the code. Does anyone have any insights on how to resolve this issue?

peterfoot, 2024-04-14:
To use the macOS implementation of this library your consuming project needs to use a mac target framework. Because of build issues we removed the maccatalyst target from the project but I'm working on adding these back in. In this case the library uses the CoreBluetooth framework (essentially the same code as iOS uses) as macOS doesn't use BlueZ.

starteNCS, 2024-04-23:
Can you expand on the "mac target framework"? Is this something we can do for our projects in the meantime to have it still work?

mjo151, 2025-02-03:
I am also seeing this issue on macOS. I am targeting net8.0-macos, but it looks like InTheHand.BluetoothLE.csproj does not include net8.0-macos in the list of TFMs.

mjo151, 2025-04-11:
@peterfoot Is it possible to add net8.0-macos to the list of TFMs in the BLE project? I tested this locally and it worked well with macOS. Thank you for your work on this project - much appreciated.