inthehand/32feet #415: Unity BluetoothBLE: Operation is not supported on this platform

erb13020, 2024-04-24:
Unity Version: 2022.3.19f1
Package: InTheHand.BluetoothLE 4.0.36
Platform: Windows 11

I'm currently having trouble using the BLuetoothLE package with my Unity project. When I run my code I get an error when trying to scan for BLE devices using Bluetooth.ScanForDevicesAsync(), _"Operation is not supported on this platform"_. Here is the code block

```
#if UNITY_STANDALONE_WIN
using InTheHand.Bluetooth;
using System.Linq;
#endif

    private async void StartScanningForDevicesWindows()
    {
#if UNITY_STANDALONE_WIN
        var devices = await Bluetooth.ScanForDevicesAsync();
        // Rest of code
```

I've tried installing the package a couple of different ways:

1. Installing using Nuget for Unity according to [Unity example](https://gitlab.com/jcs-workspace/bluetooth/32feet.Unity.Example) while ensuring that the API Compatibility Level in Unity was .NET Framework (uses 4.x) and not .NET Standard 2.1
2. Downloading the github repo from the main branch and building the dlls using Visual Studio. I first tried the .NET Standard 2.0 InTheHand.BluetoothLE.dll, but that didn't work either. I read this `[issue ](https://github.com/inthehand/32feet/issues/214#issuecomment-1241826141)` and realized the .net standard one wouldn't work.
3. I then tried using the .net462 InTheHand.BluetoothLE.dll, however, this still yielded the same "Platform Not Supported" Error.

For what it's worth, my platform (Windows 11) does support Bluetooth and Bluetooth is enabled in both Unity and Windows.

How can I fix this issue?

peterfoot, 2024-04-25:
If you check the file properties for the dll in your output folder you can see which version has been used in the project e.g. "InTheHand.BluetoothLE-net462". Because the library uses WinRT apis it references the package Microsoft.Windows.SDK.Contracts but I'm assuming that NuGet for Unity should pull this in automatically.

erb13020, 2024-04-26:
Thank you, @peterfoot

closed 2024-04-26 by erb13020