inthehand/32feet #417: Using both InTheHand.Net.Bluetooth and InTheHand.BluetoothLE in a netX-windows application.

dotnet555, 2024-05-04:
Hi,
I have the problem that I need to use both the BluetoothClient class (from InTheHand.Net.Bluetooth) and the Bluetooth static class (from InTheHand.BluetoothLE) in a net8-windows project.

This is a rundown of my experience:

- Added the InTheHand.Net.Bluetooth package, and used the BluetoothClient class succesfully
- Added the InTheHand.BluetoothLE package, and tried to use the Bluetooth static class: got the exception "No path specified for UNIX transport"
- As suggested elsewhere, targeted my project from net8-windows to net8.0-windows10.0.22621.0
- Now the Bluetooth static class works, but the BluetoothClient class stopped working.
- ... a lot of banging the head against a brick wall...
- Got a working solution:
  - Main project targeting net8.0-windows10.0.22621.0
  - Local cloned copy of InTheHand.BluetoothLE targeting only net8.0-windows10.0.22621.0, removing all the compilation conditions and leaving only the "Windows" folder under "Platforms", and making the necessary changes to make the code use the "Windows" classes.
  - Local cloned copy of InTheHand.Net.Bluetooth targeting only net8.0-windows10.0.22621.0, removing all the compilation conditions and leaving only the "Win32" folder under "Platforms", and making the necessary changes to make the code use the "Win32" classes.

With this configuration, I got both classes (BluetoothClient and Bluetooth) to work correctly.

So, my question is: there is any way to obtain the same behavior (net80-windows application, with InTheHand.BluetoothLE using the "Windows" folder and InTheHand.Net.Bluetooth using the "Win32" folder) using NuGet packages instead of using a modified local copy of the source code?

Thanks.

nigeldun, 2025-02-06:
Ive run into the same problem trying to have both LE and Classic options in my app. 
LE seems to consider itself to be on Linux without that configuration, while Classic doesn't seem to want to work no matter what windows OS version you put in.

My app is also intended to work cross platform so it seems I will have many variations to build.

(As a slight aside I also found that a BT usb dongle I had worked with LE but not Classic which caused confusion as other machines worked, swapped it for another one and Classic worked. So it seems LE is more likely to work... maybe)

Peter any chance of getting versions of both libraries that can work for the same target?

nigeldun, 2025-02-09:
Done some poking around. For me on windows 11 Classic works with a target of .net8.0 as it includes linux and win32. LE doesn’t work for .net8.0 as it only has the linux code. Now if I target .net8.0-windows10.0… LE works as it now include the windows code, however Classic which uses windows code instead of win32 now fails to pair. No idea why. If I modify a copy of the classic code to use win32 then both work, but I have to do a removeDevice before pairing in my classic using code. No idea why I’m having these problems. 
Noticed in the LE csproj file there’s support for macos but no targetframeworks entry which seems off (and I need to build for macos) but csproj files and how things build are a mystery to me.