inthehand/32feet #418: Discover devices doesn't discover device, but I can find it via microsoft bluetooth

markat1, 2024-06-04:
I'm using InTheHand.Net.Bluetooth version 4.1.43.

I tried discovering my label machine (zebra zd621), but I'm unable to show it via:

```
  public IReadOnlyCollection<BluetoothDeviceInfo> GetNearbyDevicesViaBluetooth()
  {
      using var client = new BluetoothClient();

      return client.DiscoverDevices(3);
  }
```
I can find zebra zd621 device on my phone and  microsoft bluetooth.

[image]

I tried setting inqueryLength, but I get an error back saying `System.PlatformNotSupportedException: 'Operation is not supported on this platform.'`:

```
   public IReadOnlyCollection<BluetoothDeviceInfo> GetNearbyDevicesViaBluetooth()
   {
       using var client = new BluetoothClient();

       client.InquiryLength = TimeSpan.FromSeconds(120);

       return client.DiscoverDevices(3);
   }
```

Error:

`TimeSpan IBluetoothClient.InquiryLength { get => TimeSpan.Zero; set => throw new PlatformNotSupportedException(); }`

SydneyOwl, 2024-06-06:
maybe `zebra zd621` is a ble device so you should use InTheHand.Net.BluetoothLE instead?

markat1, 2024-06-06:
hmm, I will definately try with BluetoothLE!

If I look in the documentation for zebra zd21 it says that it can both handle bluetooth classic and bluetooth LE

[image]

In the setup I can choose between or use both classic + bluetooth LE

[image]

markat1, 2024-06-06:
hmm do you have a sample on howto use BluetoothLE? I'm not sure if the following is using Bluetooth lE.

   public IAsyncEnumerable<BluetoothDeviceInfo> GetNearbyDevicesViaBluetoothAsync()
```
   {
       using var client = new BluetoothClient();
       return client.DiscoverDevicesAsync();
   }
```

SydneyOwl, 2024-06-06:
Hello @markat1
Seems like you're not using ble since you're using `BluetoothDevice` instead of `BluetoothLEDeviceInfo`.
Here's my example:
```c#
using System;
using System.Runtime.InteropServices;
using System.Threading;
using System.Threading.Tasks;
using InTheHand.Bluetooth;
using SenhaixFreqWriter.Constants.BLE;
using SenhaixFreqWriter.Utils.BLE.Interfaces;
using SenhaixFreqWriter.Utils.Serial;

namespace SenhaixFreqWriter.Utils.BLE.Platforms.Generic;

public class GenerticShxble : IBluetooth
{

................  

    public async Task<bool> ScanForShxAsync()
    {
        var filter = new BluetoothLEScanFilter
        {
            Name = BleConst.BtnameShx8800
        };
        try
        {
            _shxDevice = await Bluetooth.RequestDeviceAsync(new RequestDeviceOptions { Filters = { filter } });
        }
        catch
        {
            var cts = new CancellationTokenSource();
            cts.CancelAfter(5000);
            var discoveredDevices = await Bluetooth.ScanForDevicesAsync(new RequestDeviceOptions
            {
                Filters = { filter }
            }, cts.Token);
            foreach (var discoveredDevice in discoveredDevices)
                if (discoveredDevice.Name.Equals(BleConst.BtnameShx8800))
                {
                    _shxDevice = discoveredDevice;
                    break;
                }
        }

        return _shxDevice != null;
    }
.......
}
```

SydneyOwl, 2024-06-06:
by the way, make sure you're using `<TargetFramework>netx.0-windows10.0.19041.0</TargetFramework>` when building on windows. see #341 for more!

markat1, 2024-06-06:
You are absolutely right @SydneyOwl ! I also tried your example and it worked! Thank you!

markat1, 2024-06-06:
You are absolutely right @SydneyOwl ! I also tried your example and it worked! Thank you!

will definately check out https://github.com/inthehand/32feet/issues/341 !

SydneyOwl, 2024-06-06:
Glad to be of help to you！😃

closed 2024-06-06 by markat1

markat1, 2024-06-06:
just one last thing @SydneyOwl - do you also have a sample of writing to a BluetoothLEDevice ?

SydneyOwl, 2024-06-06:
@markat1 
here's my sample code:
```c#
using System;
using System.Runtime.InteropServices;
using System.Threading;
using System.Threading.Tasks;
using InTheHand.Bluetooth;
using SenhaixFreqWriter.Constants.BLE;
using SenhaixFreqWriter.Utils.BLE.Interfaces;
using SenhaixFreqWriter.Utils.Serial;

namespace SenhaixFreqWriter.Utils.BLE.Platforms.Generic;

public class GenerticShxble : IBluetooth
{
    private GattCharacteristic _shxCharacteristic;
    private BluetoothDevice _shxDevice;

    private GattService _shxService;

    public Task<bool> GetBleAvailabilityAsync()
    {
       ...
    }

    public async Task<bool> ScanForShxAsync()
    {
      ....
    }

    public Task ConnectShxDeviceAsync()
    {
        return _shxDevice.Gatt.ConnectAsync();
    }

    public async Task<bool> ConnectShxRwServiceAsync()
    {
        _shxService = await _shxDevice.Gatt.GetPrimaryServiceAsync(
            BluetoothUuid.FromShortId(Convert.ToUInt16(BleConst.RwServiceUuid.ToUpper(), 16)));
        return _shxService != null;
    }

    public async Task<bool> ConnectShxRwCharacteristicAsync()
    {
        _shxCharacteristic = await _shxService.GetCharacteristicAsync(
            BluetoothUuid.FromShortId(Convert.ToUInt16(BleConst.RwCharacteristicUuid.ToUpper(), 16)));
        return _shxCharacteristic != null;
    }

    public async void RegisterSerial()
    {
        _shxCharacteristic.CharacteristicValueChanged += Characteristic_CharacteristicValueChanged;
        MySerialPort.GetInstance().BtDeviceMtu = _shxDevice.Gatt.Mtu;
        await _shxCharacteristic.StartNotificationsAsync();
        MySerialPort.GetInstance().WriteBle = _shxCharacteristic.WriteValueWithoutResponseAsync;
    }

    private void Characteristic_CharacteristicValueChanged(object sender, GattCharacteristicValueChangedEventArgs e)
    {
        foreach (var b in e.Value) MySerialPort.GetInstance().RxData.Enqueue(b);
    }

    public void Dispose()
    {
        ...
    }

}
```
For a BLE (Bluetooth Low Energy) device, there may be multiple services and characteristics. You need to find the service UUID (in my code, RwServiceUuid) and characteristic UUID(in my code, RwCharacteristicUuid) corresponding to the characteristic used for data interaction. You may be able to find this information in the device's development documentation.
then you should connect to the target characteristic of the ble device at first(in my sample, ConnectShxDeviceAsync->ConnectShxRwServiceAsync->ConnectShxRwCharacteristicAsync), then you should bind a callback function `CharacteristicValueChanged` ( in my code i put all data sent by device into a queue) then call StartNotificationsAsync,, then you can call WriteValueWithoutResponseAsync or WriteValueAsync ( depends on your device）to write data to the device!

markat1, 2024-06-06:
Hi @SydneyOwl  - I hope it's okay I'm contacting you again - I believe I'm pretty close to make it work (thanks to your help help of course)! But I get an error regarding range.

So I have the following setup:
```
public async Task ConnectToZd621()
{

    BluetoothDevice? labelDevice = (await _bluetoothService.GetNearbyDevicesViaBluetoothAsync()).FirstOrDefault(device => device.Name == "ZD621");

    var service = (await labelDevice!.Gatt.GetPrimaryServicesAsync()).LastOrDefault(s => s.Device.Name == "ZD621");

    await labelDevice!.Gatt.ConnectAsync();

    var characterics = (await service!.GetCharacteristicsAsync()).Where(s => s.Properties.ToString() == "Indicate");

    var characteric = characterics.FirstOrDefault();

    characteric!.CharacteristicValueChanged += Characteristic_CharacteristicValueChanged;

    await characteric.StartNotificationsAsync();

    byte[] message = Encoding.ASCII.GetBytes(GetMessageByItemNumber());

    await characteric.WriteValueWithoutResponseAsync(message);
}

private void Characteristic_CharacteristicValueChanged(object? sender, GattCharacteristicValueChangedEventArgs e)
{

    var t = sender;
    var p = e;
}

private static string GetMessageByItemNumber()
{
    return @$"
                   ^XA
                   ^LL1422.650
                   ^COY,1000,0
                   ^FO0,70,2
                   ^A0N,120,88
                   ^FDTest^FS
                   ^XZ";
}
```
My data (ascii encoded) is 261 bytes.

[image]

But I get an error that says that the value does not fall within the expected range - how can that be?

[image]

SydneyOwl, 2024-06-06:
I‘m not sure whether proper characteristic is connected or not.. Here’s an example of the zebra iOS sdk: https://github.com/ZebraDevs/LinkOS-iOS-Samples/tree/ZebraPrinterBLEDemo/ZebraPrinterBLEDemo

And here’re the uuids which may possible works:
ZPRINTER_DIS_SERVICE_UUID = "0000180A-0000-1000-8000-00805F9B34FB" // Or "180A". 
ZPRINTER_SERVICE_UUID="38EB4A80-C570-11E3-9507-0002A5D5C51B"      / / TARGET SERVICE
READ_FROM_ZPRINTER_CHARACTERISTIC_UUID = "38EB4A81-C570-11E3-9507-0002A5D5C51B"  //USE THIS TO READ
WRITE_TO_ZPRINTER_CHARACTERISTIC_UUID  = "38EB4A82-C570-11E3-9507-0002A5D5C51B" //USE THIS TO WRITE

maybe you should try to connect to services and characteristics directly from the uuids?
If it doesn’t work, try using writewithresponse instead of writewithoutresponse?

markat1, 2024-06-07:
1. So first I choose ZD621 In the list of devices I get back from Bluetooth.ScanForDevicesAsync()

[image]

2. Then I choose my primary service - right now choosing index 3.
[image]
3. Then I connect to the gatt server via await labelDevice!.Gatt.ConnectAsync()
4.  Then I call service.GetCharacteristicsAsync and receives only back charactericstics with property "Indicate"
[image]
5. I set event on characteristicValueChanged and then ran characteric.StartNotificationsAsync()
6. I then get my byte message and checks if the byte range is within the limit of MTu. Mtu here is 247, so it's seems fine with 138 I suppose.
[image]
[image]
7. Lastly I run await characteric.WriteValueWithResponseAsync(message)  and gets back an error that doesn't really saying thing other than COM exception.

[image]

SydneyOwl, 2024-06-07:
This is really confusing... Have you consulted the SDK mentioned above? It seems like the Zebra printer uses two different characterics for reading and writing respectively, but it appears you're using the same one.

```c#
READ_FROM_ZPRINTER_CHARACTERISTIC_UUID = "38EB4A81-C570-11E3-9507-0002A5D5C51B" //USE THIS TO READ
WRITE_TO_ZPRINTER_CHARACTERISTIC_UUID = "38EB4A82-C570-11E3-9507-0002A5D5C51B" //USE THIS TO WRITE
```

markat1, 2024-06-07:
Hmm okay - so I guess It's better to show you what I can choose from of characteristics

[image]

SydneyOwl, 2024-06-07:
[image]

I don't quite understand what you mean; in fact, as indicated by the red circle in the diagram, the characteristic 38EB4A81 can only be used for indication, and you should bind the callback function for received data to this feature; whereas the characteristic 38EB4A82 can be used for read and write, and you should write data to this feature.

markat1, 2024-06-07:
Sorry for being a bit rookie here :) -  what confuses me is the following expression marked with a red circle -  should comment it out? 
[image]

SydneyOwl, 2024-06-07:
In fact, it's not like that... 
Could you paste function ConnectToZd621 at here?

SydneyOwl, 2024-06-07:
maybe this works
```c#
public async Task ConnectToZd621()
    {

        try
        {
            BluetoothDevice? labelDevice =
                (await _bluetoothService.GetNearbyDevicesViaBluetoothAsync()).FirstOrDefault(device =>
                    device.Name == "ZD621");
            IEnumerable<BluetoothDevice>? labelDevices = await _bluetoothService.GetNearbyDevicesViaBluetoothAsync();
            
            
            await labelDevice!.Gatt.ConnectAsync();

            var service =labelDevice!.Gatt.GetPrimaryServiceAsync(BluetoothUuid.FromGuid(Guid.Parse("38EB4A80-C570-11E3-9507-0002A5D5C51B"))).Result;

            var isConnnected = labelDevice!.Gatt.IsConnected;

            var read_characteric = service
                .GetCharacteristicAsync(BluetoothUuid.FromGuid(Guid.Parse("38EB4A81-C570-11E3-9507-0002A5D5C51B")))
                .Result;
            
            var write_characteric = service
                .GetCharacteristicAsync(BluetoothUuid.FromGuid(Guid.Parse("38EB4A82-C570-11E3-9507-0002A5D5C51B")))
                .Result;

            read_characteric!.CharacteristicValueChanged += Characteristic_CharacteristicValueChanged;

            await read_characteric.StartNotificationsAsync();

            byte[] message = Encoding.ASCII.GetBytes(GetMessageByItemNumber());

            var mtu = labelDevice.Gatt.Mtu;

            await write_characteric.WriteValueWithResponseAsync(message);
        }
        catch (Exception ex)
        {
            var t = ex;

            throw;
        }
    }
```

markat1, 2024-06-07:
Yes that worked! thank you so much @SydneyOwl !

markat1, 2024-06-07:
Okay one final question ! @SydneyOwl 

mtu is not something that I can make bigger?   247 bytes is all I got?

SydneyOwl, 2024-06-07:
congratulations！
My English is not very good, so you may not fully understand what I mean..
yes mtu is negotiated by the operation system and the device so you can't modify it

markat1, 2024-06-07:
you english are great @SydneyOwl ! Thank you so much for your help - I couldn't have done it without you!

SydneyOwl, 2024-06-07:
you're welcome!
actually zebra printer has their own sdk here:https://techdocs.zebra.com/link-os/
maybe you can directly use them

markat1, 2024-06-07:
tried that :) - doesn't work - much better using 32feet, because it seperate me from the product :)