inthehand/32feet #420: CharacteristicValueChanged  doesn't work  with indicate

markat1, 2024-06-19:
I'm trying to receive information from the gatt server via CharacteristicValueChanged, but it doesn't seem to work.

Characteristic is a "indicate", so it need client to "accept" the responses coming from the gattserver before it proceed. Correct me if I'm  wrong. 

`I'm not sure what to do with descriptor here, but maybe this is where I should do my acknowledge?`
```
  var readValueFromDescriptor = await descriptor.ReadValueAsync();
  await descriptor.WriteValueAsync(readValueFromDescriptor);

  readCharacteric!.CharacteristicValueChanged += Characteristic_CharacteristicValueChanged;
  await readCharacteric.StartNotificationsAsync();
```
Rest of the code

```
public async Task ConnectToZd621()
    {

         try
        {
            BluetoothDevice? labelDevice =
                (await _bluetoothService.GetNearbyDevicesViaBluetoothAsync()).FirstOrDefault(device =>
                    device.Name == "ZD621");
            IEnumerable<BluetoothDevice>? labelDevices = await _bluetoothService.GetNearbyDevicesViaBluetoothAsync();
            
            
            await labelDevice!.Gatt.ConnectAsync();

            var service = await labelDevice!.Gatt.GetPrimaryServiceAsync(BluetoothUuid.FromGuid(Guid.Parse("38EB4A80-C570-11E3-9507-0002A5D5C51B")));

            var isConnnected = labelDevice!.Gatt.IsConnected;

            var read_characteric = await service
                .GetCharacteristicAsync(BluetoothUuid.FromGuid(Guid.Parse("38EB4A81-C570-11E3-9507-0002A5D5C51B")));
            
            var write_characteric = await service
                .GetCharacteristicAsync(BluetoothUuid.FromGuid(Guid.Parse("38EB4A82-C570-11E3-9507-0002A5D5C51B")));

            read_characteric!.CharacteristicValueChanged += Characteristic_CharacteristicValueChanged;

            await read_characteric.StartNotificationsAsync();

            byte[] message = Encoding.ASCII.GetBytes(GetMessageByItemNumber());

            var mtu = labelDevice.Gatt.Mtu;

            await write_characteric.WriteValueWithResponseAsync(message);
        }
        catch (Exception ex)
        {
            var t = ex;

            throw;
        }
    }
```

SydneyOwl, 2024-06-22:
Hello @markat1 !
I don't quite understand the purpose of these two lines of code：
```c#
var readValueFromDescriptor = await descriptor.ReadValueAsync();
await descriptor.WriteValueAsync(readValueFromDescriptor);
```
if I am correct, it seems like you're reading something from a descriptor, then write it back again?
You don't need to explicitly read data from the Characteristic; You just need to register a callback function and wait for device notifications if Characteristic has an 'indicate' property.
by the way, what kind of data are you expecting to read from the printer? Maybe the printer doesn't send any data back at all!

markat1, 2024-06-24:
Hello @SydneyOwl ! Thank you so much for your reply!  

I was just experimenting with the descriptor, because I read somewhere that "indicate" characteristic require an "acknowledge" to the gatt server in order to receive any notification.

If it was just a "notification" characteristic, then I didn't have to do anything other than register a callback function.

callback function like this right?

    readCharacteric!.CharacteristicValueChanged += Characteristic_CharacteristicValueChanged;
 ```
   await readCharacteric.StartNotificationsAsync();
    
     public void Characteristic_CharacteristicValueChanged(object? sender, GattCharacteristicValueChangedEventArgs e)
     {

         var t = sender;
         var p = e;
     }
```
I'm not sure if ZD621 sends any data back, but question is: 

**How am I able to receive information back from ZD621, that it has been turned off via Device_GattServerDisconnected?**

Because this works:

```
 labelDevice.GattServerDisconnected += Device_GattServerDisconnected;

  private void Device_GattServerDisconnected(object? sender, EventArgs e)
  {
        var informationDisconnected = e;
  }
```

markat1, 2024-06-28:
No worries  @SydneyOwl 

I'm wondering about:

* how does Device_GattServerDisconnected work? Does it get a signal from zebra zd621? how?
* How does it know zebra zd621 is disconnected?

I'm just basically curious how it works together with GATT.

SydneyOwl, 2024-07-03:
I'm not entirely sure about the underlying principles, but polling the connection status should be feasible...However, continuously using task.run in a loop can incur significant overhead. You can blockingly call the connection method within the while loop instead.
My knowledge on this topic is not sufficient. sorry for not being helpful!

markat1, 2024-07-03:
no worries @SydneyOwl  

**I have a couple of extra question - I hope you don't mind :) ** 

So when Iwrite to my device, I can choose between WriteValueWithResponseAsync and WriteValueWithoutResponseAsync.

In the implementation, you can see that both methods doesn't return directly any value other than Task. 
Yet one is called WriteValueWithResponseAsync.

**Question**

- what does it mean "with response"?
- who's getting the response when I call WriteValueWithResponseAsync ?

**Why do you want a response?**
I want the response, in order to see if the device has successfull received my write request.

```
       /// <summary>
       /// Performs a Characteristic Value write to the Bluetooth LE device.
       /// </summary>
       /// <param name="value">The data to be written to the Bluetooth LE device.</param>
       /// <returns></returns>
       public Task WriteValueWithResponseAsync(byte[] value)
       {
           Bluetooth.ThrowOnInvalidAttributeValue(value);
           return PlatformWriteValue(value, true);
       }

       /// <summary>
       /// Performs a Characteristic Value write to the Bluetooth LE device.
       /// </summary>
       /// <param name="value">The data to be written to the Bluetooth LE device.</param>
       /// <returns></returns>
       public Task WriteValueWithoutResponseAsync(byte[] value)
       {
           Bluetooth.ThrowOnInvalidAttributeValue(value);
           return PlatformWriteValue(value, false);
       }
```

peterfoot, 2024-07-03:
WriteWithResponse doesn't mean you have to do anything on the receiving side as it's handled in the stack. The operation will require a response from the remote Bluetooth device so may take longer but it should pass through any errors rather than being a fire-and-forget approach. You need to check the Properties of the characteristic to see which methods it supports, and only use the supported write type.

markat1, 2024-07-04:
Thank you so much for your reply @peterfoot 

I have the following 4 properties I can use 

[image]

I was wondering how I can make use of Indicate in order to receive information from the device.

Could I write att commands and get reply back?

peterfoot, 2024-07-04:
If the property lists Write that means a WriteWithResponse. The library abstracts the difference between Notify and Indicate properties - you have to add an event handler for the CharacteristicValueChanged event on the Characteristic and call StartNotificationsAsync().

markat1, 2024-07-04:
@peterfoot 
Yep, did that - maybe device doesn't respond any data back?

```
 _gattCharacteristic = await service.GetCharacteristicAsync(BluetoothDeviceDto.Read) ?? throw new GattServerConnectionException($"characteristic {BluetoothDeviceDto.Read} not found");

 _gattCharacteristic!.CharacteristicValueChanged += GattCharacteristic_CharacteristicValueChanged;
 await _gattCharacteristic.StartNotificationsAsync();
```