inthehand/32feet #423: Filter on ScanForDevicesAsync or RequestDeviceAsync with Service UUID - Different results

ddurkee-badgermeter, 2024-06-24:
https://github.com/inthehand/32feet/issues/389
This above issue talks about this, but it still does not work for ScanForDevicesAsync on 4.0.36, even though in the linked issue it was marked resolved and this is implemented in 4.0.36.

Basically I work for a hardware company that manufactures devices that speak serial or bluetooth. We have already developed a MAUI mobile app to speak to these devices via BLE. Also a desktop app to speak to these devices serially. The new wanted functionality is the switch-swap between the two, Bluetooth on desktop, Serial on mobile. 

I have a Service UUID (unique for this device) that certainly connects on Android and iOS BLE connecting in our mobile app, now I'm trying to do the same in Desktop using 32Feet. I'll include my console app (replacing the unique service GUID with all 0's...) that I'm using to connect:

`Guid deviceguid = new Guid("00000000-0000-0000-0000-000000000000");
BluetoothUuid deviceuuid = BluetoothUuid.FromGuid(deviceguid);

var device = await Bluetooth.RequestDeviceAsync(new RequestDeviceOptions
{
    Filters = { new BluetoothLEScanFilter { Services = { deviceuuid } } }
});

var devices = await Bluetooth.ScanForDevicesAsync(new RequestDeviceOptions
{
    Filters = { new BluetoothLEScanFilter { Services = { deviceuuid } } }
});`

on RequestDeviceAsync -> the normal Windows bluetooth device list window appears but shows no device was found. So apparantly this filter is being applied but my device just isn't being discovered (despite this being the exact way I do it in the mobile app, set a filter on the scan using the service UUID and now we only see our company devices)

on ScanForDevicesAsync -> just returns ALL BLE devices in range despite the filter. Like the filter isn't even there... which was said to be fixed in this version from the linked issue, as mentioned before.

[image]

ddurkee-badgermeter, 2024-06-25:
More information:

After selecting my device from the RequestDeviceAsync window, I'm able to discover services/characteristics with other unique UUIDs for this Bluetooth module. So it's just an issue with RequestDeviceAsync/ScanForDevicesAsync that do not filter based on a correct UUID.

mt09sp, 2024-08-15:
I have to ask... Is this using the nuget package?

There's a whole heap of changes to the underlying Bluetooth.windows.cs implementation that enable the filters that I don't believe have been bundled up yet. 

The commit is back in march 2024.

peterfoot, 2024-08-24:
Latest NuGet 4.0.37 has a number of code updates added since the last release.

endyb66, 2024-09-22:
I have the same issue here with the latest NuGet, the filter seems to get applied, but the list within the device chooser is empty. Is there any fix we can work around?