inthehand/32feet #430: .Net 8 application on linux throws a not supported on this platform

nigeldun, 2024-08-21:
Hi
   I’m trying to run a .net8 app which I’ve created that works on windows on a raspberry pi running linux (it’s  actually a distribution called stellarmate used in astrophotography), however discovery seems to never return (the linux command bluetoothctl (?) lists the device I want to connect to so it is discoverable from the pi) and when I try to connect and it tries to pair/authentication it fails with operation is not supported on this platform.

  Is .net8 actually supported at this point as I see only up to.net 7 listed on nuget? If it is is there something I have to do differently on linux or in my build to make things work?

  NB I can generate a stack trace if that helps rather than my vague description.

 Regards

   Nigel

mt09sp, 2024-08-22:
The implementations are very tightly tied to the version that the nuget package is saying it supports. In nuget scroll down to the dependencies and make sure that's what you're targeting.

Also be aware that the nuget package is pretty old and things like filtering devices by name prefix aren't implemented in the nuget package but are in the source.

For me I had to take the source and strip out all the non windows stuff and am working with that and have had to add a lot of missing functionality. Maybe when i'm finished I'll bundle it into a PR. An alternative might be to include the source directly as a git sub module but git is pretty flakey at managing something so complex so unless you know what you're doing it's easier to just grab the source and work with it disconnected from github.

nigeldun, 2024-08-26:
ok I’ve now installed the latest package which looks to support .net 8. Now I don’t get a not supported error. However while discovery does discover things it never completes. I’ve tried both the async and non-async versions. I’ve tried it with a cancellation token but that either doesn’t work when using a long timeout or with a short one seems to cause an exception.
If I hard code the target device address it fails to pair. If I skip that it cannot connect. Unfortunately the inbuilt bluetooth gui doesn’t seem to discover anything so I’m not sure where the problem is. Any suggestions?

mt09sp, 2024-08-28:
This is a snippet of what i'm using to connect and you can see that I still have flow paths that i've tagged as non functioning because they're not. Many of the underlying issues actually flow into the windows native api's so not all issues are due to this library.

I started with this... https://inthehand.com/2022/12/21/12-days-of-bluetooth-8-bluetooth-low-energy-in-code/

This will get you to the point of choosing a device from a filtered list and connecting to a gatt service and then connecting to some gatt characteristics. In my case they're RX and TX for serial comms with a bluetooth device.

```
        async Task PerformConnection()
        {
            log($"raceBoxRegistry.DefaultConnectedDevice {raceBoxRegistry.DefaultConnectedDevice}");
            log($"raceBoxRegistry.LastConnectedDevice {raceBoxRegistry.LastConnectedDevice}");

            this.btnConnectDisconnect.Text = "Connecting";

            if (device != null)
            {
                //
                // NONE OF THIS WORKS
                //
                //
                if (device.IsConnected)
                {
                    // We're already connected
                    log($"Device {device.Name} is known and connected.");
                }
                else
                {
                    // We're NOT connected. We don't know if this will work.
                    log($"Device {device.Name} is known but not connected. Attempting to connect");

                    // DOESN'T WORK
                    await device.PairAsync();
                    // this paired and then timed out.

                    log($"Device {device.Name} should now be paired");
                }
            }
            else
            {
                bool connectToLastDevice = false;

                string deviceName = raceBoxRegistry.GetBluetoothDevice();
                if (!string.IsNullOrWhiteSpace(deviceName) && connectToLastDevice)
                {
                    // Connect to specific device.
                }
                else
                {
                    // Scan and show device list for devices starting with race.
                    var filterPrefix = new BluetoothLEScanFilter();
                    filterPrefix.NamePrefix = "Race";
                    var optionsPrefix = new RequestDeviceOptions();
                    optionsPrefix.Filters.Add(filterPrefix);
                    device = await Bluetooth.RequestDeviceAsync(optionsPrefix);
                }
            }

            if (device == null)
            {
                MessageBox.Show($"No device.");
                return;
            }

            //
            // At this point while we have a device we don't have a connection.
            // isPaired will be false.
            // isConnected will be false
            // NativeDevice ConnectionStatus will be disconnected
            // gatt will be identified but not connected.
 
            log($"found device {device.Name} {device.Id}");

            var gatt = device.Gatt;
            if (!gatt.IsConnected)
            {
                try
                {
                    log($"Connecting to GATT Server");

                    await gatt.ConnectAsync();
                }
                catch (Exception ex)
                {

                    log(ex.ToString());
                    if (device.IsConnected)
                    {
                        // We're already connected so make sure to disconnect
                        device.Disconnect();
                        device = null;
                        // throw new Exception("We need to clean up the device");
                    }
                    return;
                }

            }

            log($"GATT Server connection complete");

            /*
            var services = await gatt.GetPrimaryServicesAsync();
            foreach ( var service in services) {
                log($"service: {service.Uuid}");
            }
            */

            bluetoothSerialService.uartService = await gatt.GetPrimaryServiceAsync(new Guid("6e400001-b5a3-f393-e0a9-e50e24dcca9e"));
            if (bluetoothSerialService.uartService == null)
            {
                return;
            }

            log($"uartService {bluetoothSerialService.uartService.Uuid}");

            IReadOnlyList<GattCharacteristic> characteristics = await bluetoothSerialService.uartService.GetCharacteristicsAsync();

            bluetoothSerialService.rxCharacteristic = await bluetoothSerialService.uartService.GetCharacteristicAsync(new Guid("6e400002-b5a3-f393-e0a9-e50e24dcca9e"));
            bluetoothSerialService.txCharacteristic = await bluetoothSerialService.uartService.GetCharacteristicAsync(new Guid("6e400003-b5a3-f393-e0a9-e50e24dcca9e"));

            log($"Found characteristics {characteristics.Count()}");
            log($"Rx Characteristic {bluetoothSerialService.rxCharacteristic.Uuid}"); // For sending commands
            log($"Tx Characteristic {bluetoothSerialService.txCharacteristic.Uuid}"); // For reading data.

            isConnected = true;
        }

```

BluetoothLE isn't simple which is surprising. If Microsoft redirected the effort that they're putting into destroying windows into their api's then this would be a lot easier.

mt09sp, 2024-08-28:
On bug in this library though is that  IsPaired doesn't work in windows.
```
        /// <summary>
        /// Returns true if the device is paired. Doesn't work. Use IsConnected instead
        /// </summary>
        /// <remarks>Supported on Windows and Android only.</remarks>
        [Obsolete("Doesn't work. Dont use. Use IsConnected instead.", true)]
        public bool IsPaired { get { return GetIsPaired(); } }

        /// <summary>
        /// Returns true if the device is connected
        /// </summary>
        /// <remarks>Supported on Windows and Android only.</remarks>

        public bool IsConnected { get { return GetIsConnected(); } }

```
I had to add an IsConnected 
where GetIsConnected is
```
        bool GetIsConnected()
        {
            return NativeDevice.ConnectionStatus == BluetoothConnectionStatus.Connected ? true : false;
        }
```
Eventually when i'm done i'll do a proper fork and PR but now i'm working from a stripped out windows only version

nigeldun, 2024-08-28:
Hi

  is that a reply meant for me? I’m using bluetooth classic rather than ble. Currently my code runs fine on one of my windows pc’s (the other seems to have bluetooth problems generally a red herring which stumped me for ages). I’m told my code also works on an intel mac so it’s only the raspberry pi with debian based linux on that I appear to be having problems on (well until someone asks me to support android or ios :) )

Unfortunately I’ve not got another linux box to try and I’m not sure if the inbuilt bluetooth on the pi is really working properly even though things are discovered.

Regards

mt09sp, 2024-08-28:
It was meant for you but I just presumed it would be BLE given near everything is these days. I don't like your chances of success though. This library is a wrapper about the native windows .net libraries and it's those libraries where any limitations will ultimately lie. You're dependent on your windows for pi having the appropriate native support.

nigeldun, 2024-08-28:
ok. I’m using classic as the usual interface to the device is via usb aka serial.
Not sure if I’ve misled you a bit when I say my code works on windows and mac but not linux/pi I don’t mean the same binary I am publishing the code for the different architectures (sorry if this is obvious just wanting to make sure we are meaning the same thing) and so I’m assuming the library has linux specific code ‘under the hood’ which was why .net 8 wasn’t working until the library was modified to publish a version for .net 8 with linux in it.

nigeldun, 2024-08-28:
my first line of my first message said I was trying to run code that works on windows but wasn’t on a raspberry pi running linux…. :(

No problems:)

DamianSuess, 2024-09-04:
Which distro/version are you running on the Raspberry PI? I apologize in advance if I missed it in the notes above.

DamianSuess, 2024-09-04:
Nevermind, @nigeldun,  I read your last post incorrectly. It was on Windows

nigeldun, 2024-09-05:
The working version was on windows. The non-working version was on a distribution of astronomy software called stellarmate which is based on debian I believe. The target device in both cases was a telescope mount.

nigeldun, 2024-11-06:
no longer get this error so closing

closed 2024-11-06 by nigeldun