inthehand/32feet #431: Visual Studio Community 2022 Version 17.11.0 failed build

mikan66, 2024-08-28:
Any tips on how to get this to build and run any of the most basic examples?

Even just this basic article would be great! 
https://medium.com/the-pragmatic-tech-review/how-to-transfer-data-via-bluetooth-in-net-a23a9d6a77f6

[image]

SydneyOwl, 2024-08-29:
Maybe you should post some detailed build output here so that we can find the reason?

mikan66, 2024-08-29:
I think this should be pulled forward to .NET 8 long term support with MAUI, and remove out of support items like Xamarin and monoandroid10.0.  For the 32feet.NET from the article, it too looks dependent on very old packages.

peterfoot, 2024-08-29:
The current package for Bluetooth Classic is InTheHand.Net.Bluetooth. There were some changes between v3.x and v4.0 but fundamentally it works the same way. The latest release (4.1.44) includes .NET 8.0 support. We're not removing the Xamarin dlls from the package yet but that is something to consider in the future if it becomes a burden to manage.

If you'd like the author of that article to update it you'll have to reach out to them directly.