inthehand/32feet #443: Cannot apply ServiceRecord

Taro7x3, 2024-11-29:
Service records created using ServiceRecordBuilder(); cannot be applied to BluetoothListener.ServiceRecord.

`
var serviceClass = new Guid("00001101-0000-1000-8000-00805F9B34FB");
var serviceName = "Virtual Printer";

var builder = new ServiceRecordBuilder();

builder.AddServiceClass(BluetoothService.SerialPort);
builder.AddServiceClass(BluetoothService.BasicPrinting);
builder.AddServiceClass(BluetoothService.PrintingStatus);
builder.ServiceName = serviceName;

builder.AddCustomAttribute(new ServiceAttribute(0x0304,
    ServiceElement.CreateNumericalServiceElement(ElementType.UInt16, 0x0001)));

ServiceRecord record = builder.ServiceRecord;

listener = new BluetoothListener(serviceClass) 
{
    ServiceRecord = record
};

var radio = BluetoothRadio.PrimaryRadio;
if (radio != null)
{
    radio.Mode = RadioMode.Discoverable;
}

listener.Start();
`

peterfoot, 2024-12-03:
Are you getting an error, or not seeing the device advertising those details?

Taro7x3, 2024-12-03:
The "ServiceRecord (BluetoothListener.ServiceRecord)" is Read Only, so I couldn't access or create instance with original service record that I built with "ServiceRecordBuilder".
Is there any way to apply custmized Service Record to BluetoothListener?

My project is based on WPF (Visual Studio 2022, .Net 9.0) and using 32feet.NET (3.5.0).

peterfoot, 2024-12-03:
I see, there is a specific constructor which accepts a service record - public BluetoothListener(Guid service, ServiceRecord sdpRecord)