inthehand/32feet #445: PlatformNotSupportedException on Win11

MyZQL, 2024-12-03:
I just started a new console program to have a test, then I met the PlatformNotSupportedException when i called the ScanForDevicesAsync() method. Here is my project info:
```
<Project Sdk="Microsoft.NET.Sdk"> // Win11 SDK 10.0.22621

  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>net6.0</TargetFramework>
    <ImplicitUsings>enable</ImplicitUsings>
    <Nullable>enable</Nullable>
  </PropertyGroup>

  <ItemGroup>
    <PackageReference Include="InTheHand.BluetoothLE" Version="4.0.37" />
  </ItemGroup>

</Project>
```

And my demo code:
```
public static void BleScanTest1()
{
    Bluetooth.AdvertisementReceived += BluetoothOnAdvertisementReceived;
    Bluetooth.ScanForDevicesAsync().GetAwaiter().GetResult();
}
```

Are any prerequisites for a console program?

gdidelot, 2025-01-17:
Hello,

I've the same issue... Have you found a work around ?

peterfoot, 2025-01-19:
In order for the correct platform implementation to be used you have to include the full targetframework - not just net6.0. The currently supported Windows platform is net7.0-windows10.0.19041 which will run on Windows 10 19041 and above. You'd have to step back to an earlier version for .NET 6.0 - Microsoft stopped supporting .NET 6.0 last November.

JLVWJ, 2025-05-15:
> 为了使用正确的平台实现，您必须包含完整的目标框架，而不仅仅是 net6.0。当前支持的 Windows 平台是 net7.0-windows10.0.19041，它将在 Windows 10 19041 及更高版本上运行。您必须退回到 .NET 6.0 的早期版本 - 微软已于去年 11 月停止支持 .NET 6.0。

does is support net9.0 web application?