inthehand/32feet #454: Cannot get RequestDeviceAsync with BluetoothLEScanFilter working

sfrase59, 2025-03-08:
I am getting my full list of devices instead of the filtered list.
the guid below is obviously not the service guid but, I know i am using the correct guid.

Guid MSG_SERVICE = Guid.Parse("XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX");
BluetoothUuid bluetoothServiceUuid = BluetoothUuid.FromGuid(MSG_SERVICE);
 BluetoothLEScanFilter bluetoothLEScanFilter = new BluetoothLEScanFilter();
 bluetoothLEScanFilter.Services.Add(bluetoothServiceUuid);
 RequestDeviceOptions requestDeviceOptions = new RequestDeviceOptions();
 requestDeviceOptions.Filters.Clear();
 requestDeviceOptions.Filters.Add(bluetoothLEScanFilter);

 devicePicked = await Bluetooth.RequestDeviceAsync(requestDeviceOptions);

I am looking for help for anyone so inclined.
Thanks in advance

peterfoot, 2025-03-09:
What platform are you using? Currently not all platforms implement filtering. The Android implementation currently uses android.bluetooth.devicepicker and this doesn't support filtering by service uuid.

sfrase59, 2025-03-09:
I have tested on Android and iOS.
On my iPhone, I only have the one device so, I cannot say whether the filtering works there or not but, as you said, on Android, it is not implemented.  Thanks for the info.