inthehand/32feet #455: How to Obtain the response value on Characteristic with write and read permissions.

underjack, 2025-03-11:
I'm working on a Bluetooth LE project (C#, Windows Forms Applications), where there is a characteristic with both Read and Write permissions.  When writing, the server can respond with an error code if the data is invalid, and with a zero otherwise.

I call 

`await characteristic.WriteValueWithResponseAsync(data)`

and afterwards, characteristic.Value contains the contents of the characteristic.  Note that the WriteValueWithResponseAsync() returns a Task, not a Task<byte[]?> like ReadValueAsync(), and how I expected it to, so that it can return the response.

How do I obtain the response from the server?

peterfoot, 2025-03-17:
Hi, the WriteValueWithResponse is really about the device responding once the operation is complete, rather than performing a fire-and-forget operation. To read the response you need to await this operation and then perform a read.

underjack, 2025-04-09:
The trick is this is a characteristic that supports reads, so when you perform a read operation, you simply get the value of the characteristic, -not- the response to the write application.

In this project, I had moved to using Windows Bluetooth libraries directly, since the customer wants to support pairing in the application, and that is not on the stable branch yet.  In doing so, I found out that it seems this is a windows limitation, since the windows version of the calls also do not return any value other than success or not, and error codes, that I can see.