inthehand/32feet #457: Bluetooth.ScanForDevices() returning empty collection on iOS.  Works fine on Android.

DanOrlovsky, 2025-03-19:
**Environment**
.Net MAUI 8.0
iPhone 14 Pro Max
iOS 18.3.2

I'm advertising a GATT service from a Raspberry pi and would like to communicate with that service from an Android & iOS app.

I'm finding the devices using:

```
var filter = new BluetoothLEScanFilter();
filter.NamePrefix = "RASPBERYPI";

var options = new RequestDeviceOptions();
options.Filters.Add(filter);
var devices = await Bluetooth.ScanForDevicesAsync(options);

```

This works fine when I debug on Android.  It's turning up empty (immediately) when I debug on iOS.

I do get back `Permission.Granted` when I request them:

```
PermissionStatus status = PermissionStatus.Unknown;
if ((await Permissions.CheckStatusAsync<Permissions.Bluetooth>()) != PermissionStatus.Granted)
{
    status = await Permissions.RequestAsync<Permissions.Bluetooth>();
}
```
And I've also added bluetoothle to the required device capabilities:

```
<key>UIRequiredDeviceCapabilities</key>
<array>
	<string>arm64</string>
        <string>BluetoothLE</string>
</array>
```

(I've also tried adding the NSBluetoothAlwaysUsageDescription key's and the UIBackgroundModes bluetooth keys).

I'm not sure what to try next.

Any help or at least direction would be greatly appreciated.

DanOrlovsky, 2025-03-20:
So, as I continue to investigate (this feels like a permissions issue) and I noticed that InTheHand provided a bluetooth permissions class:

[Permissions](https://github.com/inthehand/32feet/wiki/Permissions#net-maui)

Of course with the release of .net 8, this is marked as obsolete.  Problem is: according to Microsoft's own docs the permission object doesn't have support for iOS, listed [here](https://learn.microsoft.com/en-us/dotnet/maui/platform-integration/appmodel/permissions?view=net-maui-8.0&tabs=windows#available-permissions)

Is this the source of my issue?  At least that's a start for me.

peterfoot, 2025-03-20:
Permissions is implemented for iOS but it is essentially a no-op. You have to declare the permission in your info.plist and the system will prompt the user when the manager is first invoked. The issue with 32feet is that the current NuGet published BluetoothLE library doesn't have an implementation of ScanForDevicesAsync and returns an empty collection immediately. However there is an implementation in the main code branch. This includes filtering by service uuid, name or nameprefix. I need to create a new NuGet release.

DanOrlovsky, 2025-03-20:
I pulled down the main branch of 32Ft.  I removed the reference to the BluetoothLE nuget package and referenced the LE project directly. 

I'm experiencing the same behavior.

I can set a breakpoint in the Bluetooth.unified.cs `Bluetooth` class in the `PlatformScanForDevices` and my app hits it - but the result is all the same.

I still suspect a permissions issue.  I'm going to troubleshoot this as the PoC I'm working on depends on bluetooth connectivity.  If it's of some benefit I can keep this thread updated with what I find: unless there may be something else I'm missing? 

iOS did not present me with a permissions dialog of any sort.

peterfoot, 2025-03-22:
Are your calls to Bluetooth APIs on the UI thread? It could be that you are not seeing the permissions dialog because of this and that will prevent it from working.