inthehand/32feet #458: Unity Android

Per6, 2025-03-20:
I have already gotten it to work on windows using Bluetooth Classic `net462` however when trying to do the same for android `net7.0-android33.0` has too high a version for unitys `4.7.1` and `monoandroid10.0` also gives a compile error.
What should i use for android?

peterfoot, 2025-03-22:
I believe the monoandroid one is the version to use for Android in Unity. What is the compile error?

Per6, 2025-03-22:
the compile error is:
```
ArgumentException: The Assembly Mono.Android is referenced by InTheHand.Net.Bluetooth 
('Assets/Plugins/Android/monoandroid10.0/InTheHand.Net.Bluetooth.dll').
But the dll is not allowed to be included or could not be found.
UnityEditor.AssemblyHelper.AddReferencedAssembliesRecurse ...
```

now that i have looked around for it, the exact same issue has been brought up before in #187.
Has there still not been a solution or workaround found?

Per6, 2025-05-28:
i have found a [UnityAndroidBluetooth](https://github.com/elmirjagudin/UnityBluetooth) however, to use it in my package that references InTheHand.Net.Bluetooth i reworked it to use namespaces and functions of the same names, types and parameters. However that is still not enough to make it recognized as the InTheHand.Net.Bluetooth package.
Do i need to build it as a dll or what are the steps necessary to have it be recognized via `<PackageReference Include="InTheHand.Net.Bluetooth" Version="4.1.44" />`?

peterfoot, 2025-05-28:
There's no way to use the libraries together without some additional work. The problem will be for the InTheHand.Net.Bluetooth to detect that it is running on the Mono runtime on Android and use that implementation instead of  the standard Android implementation. I will look into this and see what is possible.

Per6, 2025-05-28:
UnityAndroudBluetooth uses a unity class, so it will have to be it's own inthehand.net.bluetooth\4.1.44\lib folder. this also comes with having to recognize the UnityEngine namespace. i thought replacing monoandroid10.0 was a good solution however i also don't know much about the Mono runtime on Android.

Per6, 2025-06-03:
i managed to do it! it doesn't have all the functionality but supports the main stuff that i use. [UnityAndroid](https://github.com/Per6/UnityAndroid-InTheHand)

closed 2025-06-03 by Per6

Per6, 2025-06-03:
@peterfoot btw does it seem possible to add unity android?