inthehand/32feet #465: Can not find any devices

harper-rhett, 2025-05-20:
I can not seem to get any Bluetooth Classic devices to show up. Everything was working fine with the legacy package 32feet.NET. This is on Windows 11, tested on two laptops.

```
using InTheHand.Net.Bluetooth;
using InTheHand.Net.Sockets;

BluetoothClient bluetoothClient = new();
bluetoothClient.InquiryLength = TimeSpan.FromSeconds(20);
bluetoothClient.Authenticate = false;
bluetoothClient.Encrypt = false;

Console.WriteLine("Start searching");

ListDevices();
ListDevicesAsync();

void ListDevices()
{
	BluetoothDeviceInfo[] devices = bluetoothClient.DiscoverDevices().ToArray();
	Console.WriteLine($"Devices found: {devices.Length}");

	foreach (BluetoothDeviceInfo device in devices)
	{
		Console.WriteLine(device.DeviceName);
	}
}

async void ListDevicesAsync()
{
	IAsyncEnumerable<BluetoothDeviceInfo> devices = bluetoothClient.DiscoverDevicesAsync();

	await foreach (BluetoothDeviceInfo device in devices)
	{
		Console.WriteLine(device.DeviceName);
	}
}
```

peterfoot, 2025-05-22:
This is probably due to the change in API if you are building for a Windows 10/11 platform - it will use the WinRT API rather than the Win32 one. Unfortunately it is less predictable. Have you tried querying with a filter on specific services?

harper-rhett, 2025-05-22:
I appreciate the swift response. This can be done using InTheHand? Or are you suggesting I use native Windows APIs? I'm also wondering if it's possible for me to package my .net 8 application with the legacy API to avoid the trouble.

Agecaltin, 2025-06-13:
@harper-rhett @peterfoot Did you have any success? I am having similar issues with the client.DiscoverDevicesAsync() on my windows 10 laptop. I want it to give me a list of all available bluetooth devices in the area. Sometimes it works, sometimes it finds none.

harper-rhett, 2025-06-25:
I ended up building an application on an older .NET version with the legacy 32feet.NET package that scans for devices, and then communicates them to other applications (i.e. my modern .NET app) using I/O ports. I've open-sourced the repository [here](https://github.com/AU-RFID/BluetoothNoBS).

peterfoot, 2025-06-25:
The Windows branch of the code (not Win32) uses DeviceInformation.FindAllAsync. The query used for this is the Windows BluetoothDevice.GetDeviceSelector which is documented as returning all Bluetooth devices but actually if you look at the contents it is exactly the same as GetDeviceSelectorForPairingState(True) so it will not scan and will return only paired devices. I've been working on a fix which improves the BluetoothDevicePicker on Windows and the DiscoverDevices and DiscoverDevicesAsync so they will return both paired and unpaired devices. The code is in GitHub now, but I haven't created a new NuGet build yet.

harper-rhett, 2025-06-26:
The reason we turned to InTheHand in the first place is due to the trouble device selectors were giving us. No matter how we configured them, 32feet.NET would always pick up on more devices.